Ultimaker/Cura #7160: CURA V4.3.0

cnfengjie, 2020-02-25:
**Application version**
4.3.0

**Platform**
win10

**Printer**
JGAURORA

**Reproduction steps**
1. change Support Z distance default value 0.1 to any bigger
2. Slice and Preview you can see on the beginning the support away from wall big distance. After several layers it recovered.

**Screenshot(s)**
[image]

**Actual results**
Can't print Support well. Keep default value 0.1 then no this issue.

**Expected results**
Change support Z distance should no impact distance between support and wall at X,Y direction, only impact distance at Z direction between support and object surface. 

**Project file**
no need I think.

**Log file**
N/A

**Additional information**
REV.4.0.0 no this issue.

mahtDFR, 2020-02-26:
We released a new version today, can you upgrade and see if the issue is fixed?
I think there is a need for a project file to invesigate further, so can you attach a 3mf? (Zip it to upload it here).

cnfengjie, 2020-02-27:
> We released a new version today, can you upgrade and see if the issue is fixed?
> I think there is a need for a project file to investigate further if that doesn't fix it, so can you attach a 3mf? (Zip it to upload it here).

Sorry could you guide me where can I download this new version? And please find attached model for your checking.

[draw0227.zip](https://github.com/Ultimaker/Cura/files/4262128/draw0227.zip)

mahtDFR, 2020-02-27:
The latest release is [here](https://github.com/Ultimaker/Cura/releases/tag/4.5.0)

I just tried this in 4.5 and got this.
[image]

The 3mf you've sent me doesn't contain your project settings, just the model. Can you please go to File > Save... and save a 3mf from there, zip, and upload as before? Then I have the same settings you've used. Thanks

cnfengjie, 2020-02-27:
> The latest release is [here](https://github.com/Ultimaker/Cura/releases/tag/4.5.0)
>
> I just tried this in 4.5 and got this.
> [image]
>
> The 3mf you've sent me doesn't contain your project settings, just [...]

Try this, as you said "go to File > Save... and save a 3mf"
[001draw0227.zip](https://github.com/Ultimaker/Cura/files/4262414/001draw0227.zip)

This file is from REV4.4.0 but the problem still there~ I'm downloading 4.5.0 will try later~

defined value is bad result
[image]

default 0.1 is good
[image]

cnfengjie, 2020-02-27:
Uh-huh~ I found profile choose ‘draft ’, 'extra fast' etc. is OK. but ‘normal’ and ‘fine’ is Not OK~~ you can verify my result~

cnfengjie, 2020-02-28:
I Unfortunately, latest version 4.5.0 has same result~

Ghostkeeper, 2020-02-28:
> Try this, as you said "go to File > Save... and save a 3mf"
[001draw0227.zip](https://github.com/Ultimaker/Cura/files/4262414/001draw0227.zip)

That link got broken. Github is having some website trouble recently!

Could you try setting the "Support Stair Step Height" setting to 0? I'm suspecting we might have a duplicate of this ticket here: https://github.com/Ultimaker/Cura/issues/6379

cnfengjie, 2020-02-28:
> That link got broken. Github is having some website trouble recently!
>
> Could you try setting the "Support Stair Step Height" setting to 0? I'm suspecting we might have a duplicate of this ticket [...]
YES! If change  "Support Stair Step Height" default value ‘0.3’ to ‘0’ then no that issue！But I don't know what's this setting means, it's “expert” setting if change to ‘0’ what will happen？
[image]

re-upload due to bad link before
[draw0228.zip](https://github.com/Ultimaker/Cura/files/4267306/draw0228.zip)

mahtDFR, 2020-02-28:
Confirmed duplicate of #6379, will discuss.

closed 2020-02-28 by mahtDFR

Ghostkeeper, 2020-02-28:
The stair step height is meant to make the support stick less to the model where it rests on the model (the bottom side of the support). Since your print doesn't have any place (as far as I can see) where the support rests on the model, it's not going to have any negative effects to set this to 0. For other prints, you can consider raising the Support Bottom Distance a little bit maybe. See also the Settings Guide article about it: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_bottom_stair_step_height.md