Ultimaker/Cura #7162: New solutions to the holes in the "bridge"(waiting for verification)//“连桥”中出现孔洞的新解决方案（待验证）

gailulun, 2020-02-25:
When using the "bridge" function, if there is a hole in the bottom of the bridge, cura will make the nozzle suspended (see Figure 1), which will cause printing failure.
//当使用“连桥”功能时，如果在桥底设计有孔洞，cura将会使喷头悬空工作（见图1），从而使打印失败。
[image]
     Similarly, when printing a bottle-shaped model, when a thicker bottle body shrinks to a thinner neck (we might as well call it "shoulder")(See Figure 2), if the angle is 90 °, The first layer of the "shoulder" will be horizontal, which will require adding support, and it will obviously be located inside the bottle, which will be difficult to remove.
//同样的，在打印瓶状模型时，在由较粗的瓶身收缩直较细的瓶颈时（我们不妨称之为“肩”）（见图2），如果角度为90°，“肩”的第一层就会是水平的，这将需要添加支撑，这个支撑显然会位于瓶内，这将会很难拆除。
[image]
There are two solutions:
     An existing solution is to ignore holes in the first few layers and start printing holes in subsequent layers. After completion, manually remove the excess "film" plugged in the bottle mouth. (See Figure 3)
//解决方案有二：
一个已有的方案是在前几层中忽略孔洞，在之后的层中再开始打印孔洞。完成后再手动将多余出来的堵在瓶口的“薄膜”捅破后移除。（见图3）
[image]
     But it still needs to remove something - what if the model is similar to a gooseneck?
//但是这还是需要移除一些东西——可是如果模型类似于鹅颈瓶呢？
     A new scheme is proposed here, which is also improved at the first level of "shoulder". (see Fig. 4) Fig. 4 shows the first layer of "shoulder", and the printing method similar to "bridge" is adopted. This solution does not require any support or removal of any material. When the first layer is printed, there is no problem. You can print as usual.
//这里提出一种新方案,同样是在”肩“的第一层做改进。（见图4）图4所画为”肩“的第一层,采用与”连桥“相似的打印方法。这种方案不需要任何支撑，也不需要移除任何材料。当第一层打印完毕后，接下来就没有问题了，可以照常打印。
[image]
     Affected by the epidemic, I can't go to school for the moment, but the 3D printer is in school, so please verify it.
//受尽其疫情影响，我暂时没法去学校，但3D打印机却在学校，所以请各位代为验证。
     As my English level is not up to standard, all English parts are machine translation plus a few corrections. Please forgive me.
//由于本人英语水平不过关，所有英语部分均为机译外加少量修正，还请海涵。（我还发现英汉翻译还是百度好用，谷歌不好使:）

cnfengjie, 2020-02-26:
没有支撑的话，感觉就是在赌博，可靠性不高。
Without support, like gambling, reliability is low.

gailulun, 2020-02-26:
The purpose is to eliminate support, because support is difficult to be completely clear in some situations, which is also the meaning of the "bridge" option
//目的就是消除支撑，因为在某些形况下支撑很难被完全清楚，这也是“桥”选项存在的意义

Ghostkeeper, 2020-02-28:
The human-corrected English is readable. Thank you! I hope you're well and safe from the Corona virus.

Starting or stopping lines in mid-air is not good for the overhang quality either. If the nozzle needs to stop, it will accelerate to a stand-still. It will ooze there, which pushes the material down. This makes the overhang worse.

A bigger problem though is the actual implementation. Your solution would work for a circular hole, but I don't know how it would work with arbitrary shapes. It's going to be something like: Start lines at equal spacing around the hole. Each line finds the shortest way towards the bridge base. Fill in the gaps with line patterns. But if the shortest way is obstructed by other holes, what would it do?

I'm seeing a lot of potential for this technique: https://www.reddit.com/r/3Dprinting/comments/en7jk9/how_to_reduce_support_material_on_3d_prints_using/

However that is also human design, not algorithmic, and wouldn't work for concave shapes.

gailulun, 2020-03-02:
I'm very glad to see your message. I'm very sorry that I'm busy in class on the Internet these days, so I didn't see the message in time.
//非常高兴看到您的留言，非常抱歉这几天我忙着在网上上课，结果没能及时看到留言.
First, all hand sketches are on the same plot layer，and the lines depict the nozzle path
//首先，所有的手绘示意图都是在同一个打印图层中的，线条描绘的是喷头路径

As for the problem of nozzle stopping or starting in the air, the solution I can think of is to make the nozzle start to move on the basis of the formed model as much as possible, stop moving in the mid air(Figure 1), and at the same time, when the movement stops, "pull back" or make the nozzle move upward along the z-axis for a short distance, both of which are existing functions in Cura
//对于喷头在空中停止或启动的问题，我能想到的解决方法是尽量使喷头在已经成型的模型基础上开始运动，在半空中停止运动(图1），同时在停止运动时“回抽”或使喷头沿Z轴方向向上移动一小段距离，这两者都是cura中已经存在的功能
[image]
For the shape of the hole, this is really a big problem. In essence, this scheme approximates the curve by the tangent of the curve, which means that the tangent (line rather than line segment) cannot enter the cavity from the geometric point of view (Fig. 2).
//    对于孔洞的形状的问题，这确实是一个大麻烦。这个方案本质上是靠曲线的切线来逼近曲线，这也就意味着从几何上来看，切线（是直线而不是线段）不能进入空腔内部（图2）。
[image]
However, in practice, more than half of the cases are the first two in Figure 2, such as circular through-hole, through-hole to hold nut, or threaded hole (as long as it is not rectangular thread，See Figure 3 for section). These common holes are OK, and their interfaces are all in line with the requirements.
//不过还好，在实际应用中超过半数的情况是图2中的前两者，比如圆型通孔，容纳螺母的通孔，或者是螺纹孔（只要不是矩形螺纹，截面见图3）。这些常见的孔都没问题，它们的界面都是符合要求的。
[image]
The most difficult part is the red area in Figure 4
//最困难的部位是图4中红色区域
[image]
I think we can only make concessions at this time. First, print the green line as the "beam", then the orange line, and then the green line (Figure 5)
//我想这时我们只能做出让步，先打印绿色的那条线作为“横梁”，再打印橙色的，接下来是绿色的（图5）
[image]

gailulun, 2020-03-02:
I think it has something to do with topology in mathematics
//我感觉这和数学中的拓扑学有些关系

gailulun, 2020-03-02:
If you find it difficult to understand the English part, you can also use the software you trust to translate the following Chinese in the English part, which is the original words.  ：）
//如果你觉得英语部分理解有困难，你也可以用你信任的软件去翻译英语部分下面的汉语，那个是原话：）

gailulun, 2020-03-02:
In the same way, I think we can also use the solution similar to figure 5 for multiple holes, first print several beams, and then print relying on the existing part of the model and beams. We had to give in and allow the beams to enter the cavity.
//同样的，我想对于多孔洞也可以采用类似图5的解决方法，先打印几条横梁，再依托模型已有部分和横梁进行打印。我们不得不做出让步，允许那几根横梁进入空腔中。

gailulun, 2020-03-03:
I'm sorry that the interface of the rectangular thread is like this, but it does not affect understanding.
//非常抱歉矩形螺纹的界面是这样的，但是不影响理解。
[image]

Ghostkeeper, 2020-03-06:
This problem seems to land on the problem of [subdividing polygons into convex sub-areas](https://stackoverflow.com/questions/9468516/subdividing-a-general-polygon-in-a-number-of-small-convex-polygons) then.

I'll not give you too much hope. I think this problem is pretty complex. And there is a risk that it won't work well after we've implemented it.

gailulun, 2020-03-16:
I found that we can learn from. STL files. The surface of the model is approximated by triangles. When you print each triangle, you only need to find the support of three vertices.
//我发现我们可以借鉴.stl文件。由三角形近似模型表面，打印每个三角形时仅需找到三个顶点的支撑。
[image]
It is easy to find the prominent vertex ,and also the free point from the above figure, as long as you provide the attachment to that point.
//由上图很容易发现突出的那个顶点，同时也是游离的那个点，只要给那个点提供依附就可以了。
[image]
[image]
The green line is the beam.
//绿色的线是横梁.

gailulun, 2020-03-16:
After the three vertices are fixed, the three sides of the triangle are printed out and then the entire face is filled. This can be taken as a general idea
//将三个顶点固定后，再将三角形三个边打印出来，然后填充出整个面。这可以当作一个大体的思路。
[image]
As far as possible, the free point should be concentrated at the place with the greatest curvature.
//应该尽可能地将游离点集中在曲率最大的地方。
[image]
[image]
The black line in the picture is the beam
//图中黑线为横梁
The black line in the figure is the crossbeam. The crossbeam and the red / green line on its extension line are printed out in the same time, which is the single straight-line operation of the nozzle instead of twice.
//图中黑线为横梁，横梁与在其延长线上的红/绿线条为统一次打印出来的，为喷头单次直线运行，而不是两次。

gailulun, 2020-03-21:
The code is manually modified. Originally, the row spacing of the mesh was 0.4mm, but it was found that the workload was too large, so it was changed to 4mm. There are three bugs, all of which are that the extruder does not work, but basically does not affect the printing. Since I still don't have a 3D printer, I put the files here. The model is relatively large, which requires a 30 m diameter of 1.75 mm PLA material. The hot bed temperature is 60 ℃, the nozzle temperature is 210 ℃, and the floor height is 0.2 mm. Designed for ender3.
//代码为手动修改，本想使网状面的行间距为0.4mm，但发现工作量太大，就改成了4mm。其中有3处bug，均为挤出机不工作，但基本上不影响打印。鉴于我现在还是没有3d打印机，故将文件放在这里。模型比较大，要耗费30m直径为1.75mm的pla材料，热床温度为60摄氏度，喷头为210摄氏度，层高为0.2mm。为ender3设计。
The changes focus on lines 462841-464298.
Please review the code before printing.
[CFFFP_Specimen.zip](https://github.com/Ultimaker/Cura/files/4362728/CFFFP_Specimen.zip)
Mesh CAD design image.
[image]
CAD file:
[CADfile.zip](https://github.com/Ultimaker/Cura/files/4362740/CADfile.zip)

gailulun, 2023-01-27:
This model is different from the one uploaded before, I re-edited the gcode to make the spacing of the paths reasonable, I have to say that manually editing the Gcode is a frustrating job.

closed 2024-02-10 by github-actions[bot]