Ultimaker/Cura #7163: differences between inserted object and preview displayed

battistipaolo, 2020-02-26:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

battistipaolo, 2020-02-26:
[image]
[image]

closed 2020-02-26 by battistipaolo

Ghostkeeper, 2020-02-28:
Did you intend to close this issue? Did you find a solution?

If not, I suggest looking at whether the walls you're trying to print are not too thin to be printed with your line width.

Ghostkeeper, 2020-02-28:
Here is an article with several possible causes for missing parts in your print: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/missing_parts.md

battistipaolo, 2020-02-28:
Hello,
solved the problem, if the thickness of the loop is smaller than the diameter of the nozzle does not preview well.example: loop shell of .39mm. extruder of .40mm.
result: layers are not displayed well.
paolo battisti