Ultimaker/Cura #7165: [4.5.0] Saving a default profile as a custom profile causes a line width issue

CedricE11, 2020-02-26:
**Application version**
4.5.0

**Platform**
Windows 10 version 10.0.17763

**Printer**
Creality CR10 mini

**Reproduction steps**
1. Printing something with the "Normal - 0.15 mm" profile.
2. Printing something after saving the "Normal - 0.15 mm" profile as a custom profile. 

**Screenshot(s)**
n/a

**Actual results**
After step 1, there is no issue when printing.
After step 2, printing has an issue where line thickness is narrow to the point that the lines do not overlap. I did not let it run beyond the creation of a brim, so this might be specific to brim printing.

**Expected results**
There should be no difference between the default 0.15mm profile and this same profile when saved as a custom profile.

**Project file**
n/a?

**Log file**
n/a?

**Additional information**
n/a

mahtDFR, 2020-02-26:
Is this the profile that ships with Cura 4.5 for the Creatily CR10mini? Beause I can't find a "Normal profile" as you describe. Here's what I see: 
[image]

Are you using a custom profile here? Can you export and send it if so?

CedricE11, 2020-02-26:
I can't remember how I got these default profiles. Here are my options:

[image]

In Preferences -> Profiles, when I select a default profile, there is no option to export it. However, I duplicated this profile (like if I was creating a custom profile) and I included it here. Since I duplicated this default profile, if I were to use it, it would have the line width issue.
[exported default profile.zip](https://github.com/Ultimaker/Cura/files/4256815/exported.default.profile.zip)

Ghostkeeper, 2020-02-28:
That profile can't be imported in Cura 4.5 since there is no quality level "fast" for the CR-10 Mini. It seems to be made for a Custom FFF Printer or a printer without specific profiles for the printer as well.

Maybe you can share a project file that reproduces the problem? That would be under File -> Save... That should allow us to reproduce the issue exactly.

CedricE11, 2020-02-28:
@Ghostkeeper I just looked into the setup instruction docs (from Creality) that I used when setting my up CR-10 Mini a year ago. In the Cura first time run wizard, at the "select your machine" step, I would have selected "other...", and at the "other machine information" step, I would have selected "Prusa...". Note that I've been using this CR-10 Mini printer and Cura frequently for the past year with profiles derived from the default profiles, and it's only recently that I've had the issue described above. I attached a project file like you requested. I hope it helps with troubleshooting.
[7165 troubleshooting 1.zip](https://github.com/Ultimaker/Cura/files/4267605/7165.troubleshooting.1.zip)

CedricE11, 2020-02-29:
I just did some more troubleshooting; I tested the issue with several parts and several permutations of slicer setting changes leading up to the saving of a default profile as a custom profile. I observed that the issue only occured for the attached part. Again, the issue is that when I save a default profile as a custom profile, the printed lines become far too thin. I tested another part with exactly the same slicer setting changes leading up to saving the "Normal - 0.15mm" default profile as a custom profile (the changes were adding support and altering the retraction speed and distance), and the issue wasn't encountered - the printing was normal and as expected. So the issue may be specific to the attached part. Perhaps this bug is triggered by some characteristic of the attached part.
[Base member.zip](https://github.com/Ultimaker/Cura/files/4270409/Base.member.zip)

Ghostkeeper, 2020-03-03:
> @Ghostkeeper I just looked into the setup instruction docs (from Creality) that I used when setting my up CR-10 Mini a year ago. In the Cura first time run wizard, at the "select your machine" step, [...]

The base profiles for Creality changed in version 4.1, so that's most likely when this changed. This was summer 2019, less than a year ago indeed.

Ghostkeeper, 2020-03-03:
I've sliced this model using your Custom3 profile (from the .3mf file) as well as using the "Normal" profile. However the resulting g-codes are identical: [custom3-vs-normal.zip](https://github.com/Ultimaker/Cura/files/4281337/custom3-vs-normal.zip)

Could it be that other circumstances around your printer made it have a line width problem? Just randomly?

CedricE11, 2020-03-03:
Can you try saving a default profile as a custom profile, and then slicing the problematic model again with that saved profile?

Ghostkeeper, 2020-03-05:
So these are my reproduction steps:

1. Load in your 3MF file and the "Base member" STL.
2. Slice the object using the "Normal" profile (upon which the "Custom3" profile was based). Save (as test.gcode).
3. In the profile manager, duplicate the "Normal" profile.
4. Activate the new "Normal #2" profile.
5. Slice again and save (as test2.gcode).
6. Compare test.gcode with test2.gcode using KDiff.

They are still identical according to KDiff:
[image]

CedricE11, 2020-03-12:
@Ghostkeeper By duplicate the normal profile and activate the duplicated profile, do you mean that you selected the option to "Create profile from current settings/override ..." as shown here?
[image]

Ghostkeeper, 2020-03-17:
No, I mean that you'd go into "Manage Profiles..." and click on the "Duplicate" button there:

[image]

CedricE11, 2020-03-17:
I haven't done that. What I described in my last message is exactly what led to the error. Can you try reproducing the bug again but by selecting the option to "Create profile from current settings/override ..."?

Ghostkeeper, 2020-03-17:
You can't click on "Create profile from current settings/overrides..." unless you changed any settings. If you change any settings, the g-code is supposed to be different!

CedricE11, 2020-03-17:
When I'm in a default profile and I change a slicer setting, the printing still works fine. If I then save that profile without changing any more settings, the error occurs. So you can maybe compare these two scenarios for g-code differences.

Ghostkeeper, 2020-03-17:
All right, so my new reproduction steps are then:

1. Load your .3MF file to get your configuration.
2. Load the "base member" stl.
3. Switch to the Normal Quality profile to emulate you before you made your custom profile.
4. Set the line width to 0.2mm.
5. Slice and save the g-code as "test.gcode".
6. Click on "Create custom profile from current overrides..." and use the name "Custom4".
7. Slice and save the g-code as "test2.gcode".
8. Compare the g-codes in a diff viewer.

The files are now different because Cura shows the custom profile in a comment at the very end of the g-code:
[image]
The rest of the g-code is still identical.

Can you write down your exact steps to reproduce the issue, similar to what I did?

CedricE11, 2020-03-17:
I just did a bunch more testing and I think I understand the situation. This time, I found that the poor printing was occuring for the default profile (Normal 0.15mm) before I even saved it as a custom profile. And this time, I noticed that the filament flow rate was not keeping up with the commands to the filament feeder; in other words, the serated wheel that grabs the filament was spinning a bit on the filament. I guess that the last time I did testing and reported a bug, it must of been coincidence that the poor printing only occured after I saved a profile. All of this is kind of bewildering because I clearly remember seeing a pattern and I went back and forth several times when I originally observed the behavior. I'm sorry because it seems like this was a waste of your time. I get fine printing after I change the settings to increase the hotend temperature and decrease travel speeds, etc. I'm now looking into buying new filament feeder parts because it appears mine don't grab the filament well enough.

Ghostkeeper, 2020-03-19:
> All of this is kind of bewildering because I clearly remember seeing a pattern and I went back and forth several times when I originally observed the behavior.

That's the thing with randomness, you can never be sure 😄 

In the Settings Guide project I've written down some things you can tweak in your settings to prevent the feeder from eating your filament: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/unfinished_print.md#filament-grinding I hope that'll help.

closed 2020-03-19 by Ghostkeeper

CedricE11, 2020-03-19:
Thanks for the link! The info is helpful. Although to be clear, the issue wasn't that the filament was being ground but rather that the extruder wasn't grabbing fresh filament well enough.