Ultimaker/Cura #7166: Skirt and brim simultaneously

agermanltk, 2020-02-27:
I think a lot of people would benefit from combining these two features. Priming the nozzle is a function of both but the brim is more for bed adhesion improvements. If the skirt primes the nozzle first then when the brim begins to print it should lay filament down smooth and reduce the probability of failure for prints.

What I have seen in my experience is that the brim attempts to print but underextrusion at the start causes the filament to come out a bit wonky. This can interfere with other lines closer to the model creating a snowball effect ultimately failing the print. This could be avoided if the nozzle was primed prior to beginning the brim by printing a one or two line skirt. 

I’ve seen this is especially common on smaller prints. I guess due to the brim being smalle and more susceptible to the snowball effect I mentioned before
 The T. rex jaw scaled to 25% is a good representation of this issue. The jaw has a small surface are contacting the build plate which benefits from a brim. Due to not being primed the brim had imperfections and knocked off the first side of the jaw to be printed but not the second because it was primed by the time it got there

Thank you for your time

Ghostkeeper, 2020-02-28:
Alternatively, you could also prime in your start g-code instead with a line in the corner. Creality does this. Seems to be fairly effective.

From what I've seen, the outer lines of the brim tend to curl up more because of bad adhesion than because of underextrusion. When it's just a single line it'll curl up because it has not enough contact surface with the build plate yet. You might still have the problem then even if you primed with a skirt. You can test this by having a prime tower with a brim, and then setting the bed adhesion to use a skirt.

We'll discuss this with the team nevertheless, see what my colleagues think.

mahtDFR, 2020-03-03:
After a discussion we decided this is not a priority for us right now, so it's deferred.

closed 2023-07-13 by github-actions[bot]