Ultimaker/Cura #7168: Allow "Save to File" to be the default (instead of "Save to Removable")

mjskier, 2020-02-27:
**Is your feature request related to a problem? Please describe.**

I have a Western Digital My Book attached to my machine.
Every time I start Cura and slice a model, the default saving location for the .gcode file is **Save to Removable**. I need to click on the down arrow and select **Save to File**

**Describe the solution you'd like**

I would like to be able to save a setting so that the default is **Save to File** even if I have removable medias on the system.

**Describe alternatives you've considered**

I can use File->Export, or I can select the down arrow and select "Save to File". But it is annoying when that save dialog is obviously meant as a convenience.

**Affected users and/or printers**

Anybody who has removable devices attached to the machine when Cura is started and don't need to write the sliced model to a SD or other removable media. So pretty much anybody who use Octoprint or similar interface to their printer.

**Additional context**
[image]

leothehuman, 2020-02-27:
Second this request!

smartavionics, 2020-02-27:
As a workaround, you can just disable the removable drive support in the Marketplace dialog...

[image]

nallath, 2020-02-27:
Anyone using octoprint would be better off installing the octoprint plugin, since it enables an even easier flow; Directly sending it to octoprint.

Anyhow, I will discuss the feature request with the rest of the team.

mjskier, 2020-02-27:
Thank you for the quick replies. I will try the suggestions.
The most user friendly behavior would be for Cura to remember the last choice over exit and restart.

mjskier, 2020-02-28:
The workaround from @smartavionics works great for me.
So I'm going to close this issue. If somebody feels strongly about it I assume they can just reopen it.

fieldOfView, 2020-02-28:
> So I'm going to close this issue. If somebody feels strongly about it I assume they can just reopen it.

Please don't. I feel strongly about it, but I cannot reopen the issue. I agree with you that the most user friendly behavior would be for Cura to remember the last choice over exit and restart.

fieldOfView, 2020-02-28:
I have a PR here: https://github.com/Ultimaker/Uranium/pull/585. This remembers the last used output device between sessions.

mahtDFR, 2020-03-02:
We have an internal ticket for this: CURA-7269

fieldOfView, 2020-12-04:
Cura now remembers the last used option, so I think this can be closed.

mjskier, 2020-12-05:
I don't know, the code was merged on April 1st. It could be a joke... ;-)

closed 2020-12-05 by mjskier