Ultimaker/Cura #7169: Cura 4.5.0 fails in Tests

ggardet, 2020-02-27:
**Application version**
4.5.0

**Platform**
x86_64 (and aarch64)

**Printer**
N/A

**Reproduction steps**
Try to build Cura 4.5.0 on opeNSUSE Tumbleweed or openSUSE Leap 15.2.

**Screenshot(s)**
N/A

**Actual results**
Test fails.

**Expected results**
Test should pass.

**Project file**
N/A

**Log file**
Tests log:
```
[   92s] + make CTEST_OUTPUT_ON_FAILURE=TRUE test
[   92s] Running tests...
[   92s] /usr/bin/ctest --force-new-ctest-process 
[   92s] Test project /home/abuild/rpmbuild/BUILD/Cura-4.5.0/build
[   92s]       Start  1: invalid-imports
[   92s]  1/12 Test  #1: invalid-imports ..................   Passed    0.24 sec
[   92s]       Start  2: pytest-main
[  140s]  2/12 Test  #2: pytest-main ......................***Failed   48.15 sec
[  140s] ============================= test session starts ==============================
[  140s] platform linux -- Python 3.7.3, pytest-5.3.5, py-1.8.1, pluggy-0.13.1
[  140s] rootdir: /home/abuild/rpmbuild/BUILD/Cura-4.5.0, inifile: pytest.ini
[  140s] collected 3663 items
[  140s] 
[  140s] ../tests/TestArrange.py ...........................                      [  0%]
[  140s] ../tests/TestBuildVolume.py ....................................         [  1%]
[  140s] ../tests/TestConvexHullDecorator.py ..............................       [  2%]
[  140s] ../tests/TestCuraSceneController.py ....                                 [  2%]
[  140s] ../tests/TestCuraSceneNode.py .....                                      [  2%]
[  140s] ../tests/TestExtruderManager.py ..                                       [  2%]
[  140s] ../tests/TestGCodeListDecorator.py ..                                    [  2%]
[  140s] ../tests/TestIntentManager.py ....                                       [  3%]
[  140s] ../tests/TestLayer.py ....                                               [  3%]
[  140s] ../tests/TestMachineAction.py ...                                        [  3%]
[  140s] ../tests/TestMachineManager.py s......                                   [  3%]
[  140s] ../tests/TestOAuth2.py ..............                                    [  3%]
[  140s] ../tests/TestObjectsModel.py .................                           [  4%]
[  140s] ../tests/TestPrintInformation.py ....                                    [  4%]
[  140s] ../tests/TestProfileRequirements.py .....                                [  4%]
[  140s] ../tests/API/TestAccount.py .......                                      [  4%]
[  140s] ../tests/Machines/TestContainerTree.py .....                             [  4%]
[  140s] ../tests/Machines/TestMachineNode.py ....                                [  4%]
[  140s] ../tests/Machines/TestMaterialNode.py ......                             [  5%]
[  140s] ../tests/Machines/TestQualityNode.py .                                   [  5%]
[  140s] ../tests/Machines/TestVariantNode.py ................                    [  5%]
[  140s] ../tests/Machines/Models/TestDiscoveredPrintersModel.py s...             [  5%]
[  140s] ../tests/PrinterOutput/TestNetworkedPrinterOutputDevice.py .......       [  5%]
[  140s] ../tests/PrinterOutput/TestPrinterOutputDevice.py ....                   [  5%]
[  140s] ../tests/PrinterOutput/Models/TestPrintJobOutputModel.py .........       [  6%]
[  140s] ../tests/PrinterOutput/Models/TestPrinterConfigurationModel.py ...       [  6%]
[  140s] ../tests/PrinterOutput/Models/TestPrinterOutputModel.py ................ [  6%]
[  140s] .                                                                        [  6%]
[  140s] ../tests/Settings/TestContainerManager.py .......                        [  6%]
[  140s] ../tests/Settings/TestCuraContainerRegistry.py .....................s... [  7%]
[... 174 lines cut ...]
[  145s]       Start  7: pytest-VersionUpgrade27to30
[  145s]  7/12 Test  #7: pytest-VersionUpgrade27to30 ......   Passed    0.66 sec
[  145s]       Start  8: pytest-VersionUpgrade34to35
[  146s]  8/12 Test  #8: pytest-VersionUpgrade34to35 ......   Passed    0.61 sec
[  146s]       Start  9: pytest-VersionUpgrade43to44
[  147s]  9/12 Test  #9: pytest-VersionUpgrade43to44 ......   Passed    0.60 sec
[  147s]       Start 10: pytest-VersionUpgrade44to45
[  147s] 10/12 Test #10: pytest-VersionUpgrade44to45 ......   Passed    0.66 sec
[  147s]       Start 11: pytest-XmlMaterialProfile
[  149s] 11/12 Test #11: pytest-XmlMaterialProfile ........   Passed    2.20 sec
[  149s]       Start 12: shortcut-keys
[  150s] 12/12 Test #12: shortcut-keys ....................   Passed    0.47 sec
[  150s] 
[  150s] 92% tests passed, 1 tests failed out of 12
[  150s] 
[  150s] Total Test time (real) =  57.97 sec
[  150s] 
[  150s] The following tests FAILED:
[  150s]          2 - pytest-main (Failed)
[  150s] Errors while running CTest
[  150s] make: *** [Makefile:86: test] Error 8
```

**Additional information**
N/A

nallath, 2020-02-27:
Uh, that's weird. We run our CI with every commit, and they are succeeding right now. That's also done on Linux.

Ghostkeeper, 2020-03-02:
Are you sure that Uranium is up-to-date?

closed 2024-12-07 by github-actions[bot]