Ultimaker/Cura #7170: Cannot slice when using one at a time option when model is rather wide

alpha8196, 2020-02-27:
**Application version**
4.4.1

**Platform**
Win10 , 960M

**Printer**
I3 clone, not linked to Cura 

**Reproduction steps**
slice with all at a time, everything is fine
slice with one at a time, failed with super big grey area
also same result if start with one at a time selected

if the object is small enough it would be fine, but when it is large enough, 
even with enough space for the extruder, it would still fail to slice

**Screenshot(s)**

before
[image]

after
[image]

**Actual results**
unable to slice
**Expected results**
Slicing
**Project file**

[Bed_Leveling_Test.zip](https://github.com/Ultimaker/Cura/files/4262534/Bed_Leveling_Test.zip)

**Log file**

2020-02-27 23:45:18,041 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2020-02-27 23:45:18,042 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2020-02-27 23:45:18,048 - WARNING - [Thread-7] UM.Decorators.deprecated_function [19]: <function GlobalStack.extruders at 0x00000211BB43ED08> is deprecated (since 4.4): Please use extruderList instead.
2020-02-27 23:45:18,624 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [429]: Sending slice message took 0.5814735889434814 seconds
2020-02-27 23:45:18,776 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING]  -s raft_base_acceleration="3000" -s switch_extruder_retraction_speeds="20" -s raft_interface_line_spacing="1.0" -s roofing_extruder_nr="-1" -s support_extruder_nr="0" -s speed_support_interface="40.0" -s prime_tower_brim_enable="False" -s raft_surface_fan_speed="0" -s jerk_skirt_brim="20" -s jerk_wall="20" -s jerk_infill="20" -s skin_material_flow="100" -s bridge_skin_speed_2="15.0" -s support_z_distance="0.1" -s travel_compensate_overlapping_walls_enabled="True" -s top_layers="6" -s skirt_brim_minimal_length="250" -s switch_extruder_retraction_amount="16" -s bottom_skin_preshrink="0.8" -s cross_support_density_image="" -s wireframe_flow_connection="100" -s support_roof_enable="False" -s gradual_support_infill_steps="0" -s mold_angle="40" -s wipe_hop_enable="True" -s small_feature_speed_factor="50" -s wall_0_wipe_dist="0.2" -s wipe_retraction_extra_prime_amount="0" -s skin_angles="[ ]" -s material_flush_purge_length="60" -s center_object="False" -s skin_line_width="0.4" -s switch_extruder_retraction_speed="20" -s extruder_prime_pos_z="0" -s support_conical_min_width="5.0" -s retraction_prime_speed="25" -s extruder_prime_pos_x="0" -s raft_surface_line_width="0.4" -s acceleration_travel_layer_0="5000.0" -s mold_roof_height="0.5" -s machine_max_acceleration_z="100" -s acceleration_wall_0="3000" -s raft_speed="30.0" -s support_infill_rate="15" -s adhesion_type="skirt" -s raft_base_line_width="0.8" -s speed="0" -s wall_thickness="0.8" -s machine_height="200" -s min_infill_area="0" -s mesh_position_z="0" -s wireframe_flow_flat="100" -s magic_fuzzy_skin_enabled="False" -s retraction_hop_only_when_collides="False" -s speed_z_hop="10" -s wall_overhang_angle="90" -s z_seam_relative="False" -s support_interface_skip_height="0.3" -s brim_width="8.0" -s speed_support="60" -s skin_no_small_gaps_heuristic="False" -s material_flow="100" -s support_mesh="False" -s retraction_hop_after_extruder_switch_height="1" -s wipe_hop_speed="100" -s infill_randomize_start_location="False" -s raft_fan_speed="0" -s machine_gcode_flavor="RepRap (Marlin/Sprinter)" -s support_roof_line_width="0.6" -s spaghetti_max_infill_angle="10" -s jerk_wall_x="20" -s material_diameter="2.85" -s skin_outline_count="1" -s retraction_enable="True" -s speed_wall_0="30.0" -s support_connect_zigzags="True" -s wipe_pause="0" -s wireframe_flow="100" -s support_bottom_extruder_nr="0" -s jerk_travel_layer_0="30.0" -s support_tree_angle="40" -s support_bottom_stair_step_height="0.3" -s connect_skin_polygons="False" -s draft_shield_height_limitation="full" -s machine_max_jerk_e="5.0" -s meshfix="0" -s outer_inset_first="False" -s adhesion_extruder_nr="0" -s machine_max_feedrate_x="299792458000" -s raft_base_speed="22.5" -s support_offset="0" -s acceleration_topbottom="3000" -s slicing_tolerance="middle" -s machine_max_acceleration_x="9000" -s wall_0_material_flow="100" -s support_brim_line_count="20" -s raft_surface_acceleration="3000" -s support_roof_line_distance="0.6" -s support_tree_branch_diameter="2" -s support_tree_enable="False" -s support_roof_density="100" -s draft_shield_height="10" -s bridge_skin_density="100" -s bridge_wall_coast="100" -s material_print_temperature="210" -s support_extruder_nr_layer_0="0" -s speed_support_bottom="40.0" -s material_break_preparation_retracted_position="-16" -s xy_offset="0" -s carve_multiple_volumes="False" -s skin_overlap_mm="0.02" -s support_roof_offset="0.0" -s wireframe_top_delay="0" -s speed_support_roof="40.0" -s ironing_pattern="zigzag" -s jerk_layer_0="20" -s layer_height_0="0.2" -s bridge_skin_material_flow_3="110" -s speed_infill="60" -s support_roof_height="1" -s retraction_speed="25" -s speed_equalize_flow_enabled="False" -s skirt_brim_material_flow="100" -s build_volume_temperature="28" -s machine_min_cool_heat_time_window="50.0" -s default_material_print_temperature="210" -s wireframe_fall_down="0.5" -s support_interface_extruder_nr="0" -s spaghetti_infill_extra_volume="0" -s infill_pattern="grid" -s resolution="0" -s machine_show_variants="False" -s machine_max_acceleration_y="9000" -s wipe_retraction_speed="5" -s cross_infill_pocket_size="4.0" -s material_break_preparation_speed="2" -s meshfix_union_all="True" -s support_fan_enable="False" -s top_thickness="0.8" -s travel_compensate_overlapping_walls_0_enabled="True" -s infill_angles="[ ]" -s raft_interface_jerk="20" -s wireframe_printspeed="5" -s minimum_interface_area="1.0" -s gradual_infill_steps="0" -s adaptive_layer_height_enabled="False" -s support_tower_roof_angle="65" -s raft_acceleration="3000" -s wireframe_height="3" -s small_hole_max_size="0" -s raft_base_line_spacing="1.6" -s support_pattern="zigzag" -s wipe_hop_amount="1" -s travel_avoid_other_parts="True" -s time="23:45:18" -s meshfix_maximum_travel_resolution="0.8" -s wireframe_bottom_delay="0" -s gantry_height="55" -s top_skin_preshrink="0.8" -s extruder_prime_pos_abs="False" -s material_print_temperature_layer_0="210" -s support_interface_enable="False" -s wipe_move_distance="20" -s infill_material_flow="100" -s machine_filament_park_distance="16" -s bridge_skin_material_flow_2="100" -s support_conical_enabled="False" -s machine_nozzle_head_distance="3" -s material_print_temp_wait="True" -s wall_line_count="2" -s support_bottom_material_flow="100" -s acceleration_layer_0="3000" -s machine_start_gcode="G21 ;metric values
2020-02-27 23:45:18,780 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G90 ;absolute positioning
2020-02-27 23:45:18,782 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] M82 ;set extruder to absolute mode
2020-02-27 23:45:18,786 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] M107 ;start with the fan off
2020-02-27 23:45:18,790 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G28 X0 Y0 ;move X/Y to min endstops
2020-02-27 23:45:18,793 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G28 Z0 ;move Z to min endstops
2020-02-27 23:45:18,797 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G1 Z15.0 F9000 ;move the platform down 15mm
2020-02-27 23:45:18,800 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G92 E0 ;zero the extruded length
2020-02-27 23:45:18,803 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G1 F200 E3 ;extrude 3mm of feed stock
2020-02-27 23:45:18,807 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G92 E0 ;zero the extruded length again
2020-02-27 23:45:18,811 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] G1 F9000
2020-02-27 23:45:18,815 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] ;Put printing message on LCD screen
2020-02-27 23:45:18,819 - DEBUG - [Thread-19] UM.Backend.Backend._backendLog [101]: [Backend] M117 Printing..." -s wireframe_up_half_speed="0.3" -s jerk_roofing="20" -s support_interface_material_flow="100" -s machine_max_acceleration_e="10000" -s day="Thu" -s material_bed_temperature_layer_0="60" -s multiple_mesh_overlap="0.15" -s material_crystallinity="False" -s machine_min [... cut ...]
2020-02-27 23:45:24,962 - DEBUG - [Thread-22] UM.Backend.Backend._backendLog [101]: [Backend] GNU Affero General Public License for more details.
2020-02-27 23:45:24,964 - DEBUG - [Thread-22] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-02-27 23:45:24,967 - DEBUG - [Thread-22] UM.Backend.Backend._backendLog [101]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-02-27 23:45:24,972 - DEBUG - [Thread-22] UM.Backend.Backend._backendLog [101]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.

mahtDFR, 2020-02-28:
Can you check your machine settings, particularly 'Printhead settings'?

Ghostkeeper, 2020-03-02:
With a normal Prusa i3 in Cura it slices normally. We haven't been able to reproduce this so far.

[image]

I think this is a mistake in your printer's definition file. If you modded Cura in order to have that definition file in, a project file won't be of any use either. What is the project file?

alpha8196, 2020-03-02:
sorry for the late reply, I changed the printhead setting to be the same with @mahtDFR but still has the same issue
and sorry for misunderstanding what the project file means, here's the zipped .3mf

[PI3_Bed_Leveling_Test.zip](https://github.com/Ultimaker/Cura/files/4276541/PI3_Bed_Leveling_Test.zip)

mahtDFR, 2020-03-03:
Thanks for the project file. It sliced fine for me, so I can't reproduce the issue. As Ghostkeeper says, perhaps your definition file has been modded somehow.

Ghostkeeper, 2020-03-06:
Ah, I see what's happening. This issue reproduces in Cura 4.4 but not in Cura 4.5. We had an issue in 4.4 where one-at-a-time mode was not giving the correct collision area. That's why we couldn't reproduce it: It's been fixed in Cura 4.5.

closed 2020-03-06 by Ghostkeeper