Ultimaker/Cura #7172: [4.5.0] Pre-Heat temperatures not set

TheNitek, 2020-02-27:
I recently upgraded to 4.5.0 and now I have the problem that the pre-heat temperatures for my printer (connected via OctoPrint) are not set by default. Before these temperatures where always prefilled with the current profiles temperatures, now the fields are just empty. Once I enter the temperatures everything works, but I think the input fields should have preset values again.

fieldOfView, 2020-02-28:
> Before these temperatures where always prefilled with the current profiles temperatures, now the fields are just empty. 

I'm sorry, which fields, where? Please supply more information a screenshot.

There's a template for issues, please use it next time.

TheNitek, 2020-02-28:
[image]
As you can see those two fields are empty now. In earlier Cura versions they have been prepopulated with the currents profiles temperture settings.

Sorry for not using the template. Will do next time!

mahtDFR, 2020-02-28:
Just tried this with a UM2 over serial and it's indeed not showing values in the monitor. Will sticky it.

fieldOfView, 2020-02-28:
This is caused by https://github.com/Ultimaker/Uranium/commit/2d4dab687afff47df33c6237910bb373fc99ebaa. I created a fix here: https://github.com/Ultimaker/Cura/pull/7178

PR #7178 (2020-03-09): Fix values in print monitor preheat fields
This PR fixes the values shown in the preheat temperature fields on the Print Monitor (USB or OctoPrint). As reported in #7172, this functionality was broken in https://github.com/Ultimaker/Uranium/commit/2d4dab687afff47df33c6237910bb373fc99ebaa, but since that is a performance optimization this PR works around those changes instead of reverting them.

Fixes #7172
closed 2020-03-09 by nallath