Ultimaker/Cura #7173: Use multiple cores

controlflow, 2020-02-27:
Cura is a quite resource-insensitive app when slicing in terms of CPU usage... but it basically do not uses multiple cores at all :\ This is 16-core Ryzen CPU doing simple slicing for 30+ minutes (this is single thread working, jumping back and forth between cores + SMT):

[image]

While I understand that slicing in kinda sequential process (next layer entry point is kinda dependent on previous layer), but it seems that it can be easily done in parallel layer-by-layer (even with possible quality degradation/more stringing/non-optimal seam location), even without supports generation.

It seems that since #876 from year 2016 nothing has changed much...

smartavionics, 2020-02-27:
The part of Cura that does the actual slicing, CuraEngine, does indeed use multiple cores for a large proportion of the work it does. Some of the work has to be done serially but a lot can, and is, parallelised. Your above graphic does not reflect reality.

smartavionics, 2020-02-27:
I only have 4 cores but notice here how CuraEngine is using 357% of CPU while slicing...

[image]

mahtDFR, 2020-02-28:
I just sliced a cube at 0.002 layer height but I cant reproduce it because it's using all my cores here.
[image]

Can you share a log? That may give some clues.

controlflow, 2020-02-28:
To reproduce, slice the biggest 300x300 model from: https://www.thingiverse.com/thing:2916907
Disabling support/tree support generation speeds up things 10x times, but still takes a 5+ minutes to complete with most of the cores doing nothing.

smartavionics, 2020-02-28:
I sliced one of those chains and for most of the time, all 4 cores were busy.

Ghostkeeper, 2020-03-02:
Could you go into your task manager while Cura is running, then right-click on Cura.exe, and check "affinity"? Are all cores checked there? And the same while it's slicing and for CuraEngine.exe?

closed 2020-06-19 by no-response[bot]

v1n2e7t, 2020-07-10:
Could this be a Ryzen AMD issue? I’m seeing the same thing on my Ryzen 7 3800X

Ghostkeeper, 2020-07-13:
We have two known/reproducible bugs regarding this, mind you: There is no multithreading on MacOS at all. And with Tree Support there is multi-threading but it's largely defeated by a caching system.

I don't think it's an issue with Ryzen since I'm using a Ryzen 9 3950X myself. I also logically wouldn't think this depends on the processor; more on the operating system.

v1n2e7t, 2020-07-13:
Thank you, I am on windows but was using tree supports.

Fuzzypyro, 2020-10-20:
Can 100% confirm its not an issue between Intel/AMD. Found it's just how it processes tree supports.I ran a test between two systems just to make sure it wasn't an issue with poor optimization with infinity which in hindsight doesn't make sense anyway since it's only processing on 1 core and not say half. Hopefully they are able to find a way to split the load of tree support generation among multiple cores in the near future.

Tested on Manjaro using appimage 4.7.1

robherc, 2023-09-29:
I'm running Lubuntu on a laptop with a Ryzen7 cpu. Cura has recently (within the past week) started showing this problem (could possibly be an incompatibility with a recent OS update?).

The past 8 slices have all taken multiple minutes (possibly because of running postprocessing scripts?), and used only 13% cpu (4-core, 8-thread architecture).

fieldOfView, 2023-09-30:
Postprocessing scripts do not run until just before the gcode is saved or printed, so they do not affect slicing/previewing.

Smarti0101, 2023-11-15:
Same here on a Ryzen9 5950X (16c/32t). Max usage 5%.....

TheRealKasumi, 2023-11-20:
Exactly the same "issue" here on a Windows system with a 7950X and 128GB RAM. It seems that only 4 cores are used. Could explain why it can't be seen in the screenshot above where the CPU only has 4 cores.

eazrael, 2023-11-25:
Ryzen 7700X, Win10,, no problems, large parts use all 16 virtual cores. It really depends on the amount of parallelizable and serial work. If the parallelizable work is done fast as all cores are used, you may not even see the spike. 

Is there any debug mode or so where the slicing times are logged?