Ultimaker/Cura #7175: "News popup on startup"

azalanono, 2020-02-28:
Hello.
I always have the same troubled with every change to a newer AppImage of Cura.
The popup window opens with every startup, and also it does not remember my marketplace login.

After some time (clicking this and that, can't remember), it happens that this nasty popup vanishes and my login is memorized.

How can I manually get this done? It is really very, very nasty. This happened at least during last 3 versions of Cura, now (again) with Cura 4.5.0.

I am using openSUSE Linux Leap 15.0 x64.

mahtDFR, 2020-02-28:
Hey there - are you talking about this?
[image]

This shouldn't happen in every instance so perhaps it's specific to OpenSUSE. As a workaround:
The config files in Linux are stored in ``$USER/.local/share/cura/<Cura version>/cura.cfg``
You can manually change this value to ``latest_version_changelog_shown = 4.5.0``, this should prevent it showing. But the account login is confusing. If you check the ``cura.cfg`` file you should see ``[general]
ultimaker_auth_data =`` followed by an access token, which will be evidence of a login. If your login doesnt persist, then somehow this isn't getting saved.

azalanono, 2020-02-28:
Am Freitag, 28. Februar 2020, 10:43:03 CET schrieb maht:
> Hey there - are you talking about this?
> ![image](https://user-images.githubusercontent.com/17821912/75535510-7d3a308
> 0-5a15-11ea-9bdd-089ea68523c7.png)

Yes, exactly.
 
> This shouldn't happen in every instance so perhaps it's specific to
> OpenSUSE. As a workaround: The config files in Linux are stored in
> ``$USER/.local/share/cura/<Cura version>/cura.cfg`` You can [...]

Wouldn't it be a good idea to add a menu entry like "Store current settings"?

I think I remember how I got it solved last time.  I made a online backup and restore. This 
possibly enforced the writing of settings?

azalanono, 2020-02-28:
> This shouldn't happen in every instance so perhaps it's specific to
> OpenSUSE. 
I don't think so.

> As a workaround: The config files in Linux are stored in
> ``$USER/.local/share/cura/<Cura version>/cura.cfg`` You can manually change
> this value to ``latest_version_changelog_shown = 4.5.0``, this [...]

I had a look into cura.cnf.  You won't believe it, but there I saw "3.6.0" as version number. I 
manually changed that to 4.5.0 wherever I found it. Then Cura didn't show this popup 
anymore.

But the problem with the login persisted.
My workaround (solving this issue, too) was:
     *  Click on the connect button
     *  Firefox pops up and I get the prompt that I am already signed in.
     *  I close Firefox
     *  I open the Backup Manager and create a manual backup.
     *  I select then "restore" this recent backup.

When I start Cura again, I am automatically signed in as I expect.

Something is not working properly with migration, I guess.

mahtDFR, 2020-02-28:
Good to see you have a workaround.

> Wouldn't it be a good idea to add a menu entry like "Store current settings"?
This shouldn't be necessary as it's expected behaviour to store the settings by default. On a version upgrade, settings should get carried across from the old version, as long as a folder for a previous version exists in the ``$USER/.local/share/cura/<Cura version>/cura.cfg`` location.

If you can provide a log, it might give some clues as to why this didn't happen for you.

azalanono, 2020-02-28:
Am Freitag, 28. Februar 2020, 12:57:24 CET schrieb maht:
> Good to see you have a workaround.
>
> This shouldn't be necessary as it's expected behaviour to store the settings
> by default. On a version upgrade, settings should get carried across from
> the old [...]

I found a cura.log in the 4.5 directory.  This file you need?

I can put it on my server, of course.

mahtDFR, 2020-02-28:
Yeah that works, or you can just upload it here. This is all covered in the issue template ;)

azalanono, 2020-02-28:
[cura.log](https://github.com/Ultimaker/Cura/files/4266971/cura.log)

azalanono, 2020-02-28:
Hope this is helpful.

mahtDFR, 2020-02-28:
The most relevant snippet I can find is this:

``2020-02-28 10:53:22,987 - INFO - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for configuration file [./cura.cfg~] with type [preferences] and source version [6000009]
2020-02-28 10:53:23,027 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [309]: Failed to get file version: ../generic_petg_175+%233.xml.fdm_material, skip it
2020-02-28 10:53:23,030 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [309]: Failed to get file version: ../generic_pla_175+%232.xml.fdm_material, skip it
2020-02-28 10:53:23,045 - INFO - [MainThread] UM.VersionUpgradeManager._upgradeFile [396]: Upgraded ./cura.cfg~ to version 6000011.``

Which would suggest your cfg file is getting updated, but it's unclear to me if this is the first run since you manually edited it.

azalanono, 2020-02-28:
Am Freitag, 28. Februar 2020, 13:41:21 CET schrieb maht:
> The most relevant snippet I can find is this:
>
> ``2020-02-28 10:53:22,987 - INFO - [MainThread]
> UM.VersionUpgradeManager._getUpgradeTasks [313]: Create upgrade task for
> configuration file [...]

No, it isn't.   If the log is overwritten on every startup, I have no older details.

I edited the config file manually, replacing all 3.6.0 entries by 4.5.0.
Then I started cura to see if this helped.   It DID, the popup didn't appear again.

Then I repeated the login, made a manual backup + restore. This (restoring) automatically 
ends Cura.
Then I started Cura again and noticed that the login is now also happening automatically.

So this was at least 2 startups, but I also used Cura after that.

Ghostkeeper, 2020-03-02:
> Wouldn't it be a good idea to add a menu entry like "Store current settings"?

And where would it store the setting whether or not to store the current settings? :smile: The uninstaller has an option to erase the current settings, if necessary. Or you can delete the configuration folders that pop up when you go to Help -> Show Configuration Folder...

The log file doesn't show anything specific except that it is indeed upgrading files from an older version as if you had never started Cura 4.5 before.

Perhaps Cura is unable to write your configuration files due to access rights? Are you sure that you have write access to the files in the configuration folder? Maybe it could break due to running Cura as administrator at some point in the past?

azalanono, 2020-03-02:
Am Montag, 2. März 2020, 13:11:19 CET schrieb Ghostkeeper:
> And where would it store the setting whether or not to store the current
> settings? :smile: The uninstaller has an option to erase the current
> settings, if necessary. Or you can delete the [...]

When I click this menu item, I enforce writing the config, nothing else.

> The log file doesn't show anything specific except that it is indeed
> upgrading files from an older version as if you had never started Cura 4.5
> before.
>
> Perhaps Cura is unable to write your [...]
In this case Cura also could not create the folder "4.5" and fill in the files. But the folder and 
content exists.

I am also wondering how it can happen at all that I find "3.6.0" version numbers in the 
"4.5.0" config file. Does Cura not check for plausibility? It should do so and show an error 
if it finds something strange.

Ghostkeeper, 2020-03-03:
> I am also wondering how it can happen at all that I find "3.6.0" version numbers in the
"4.5.0" config file. Does Cura not check for plausibility? It should do so and show an error
if it finds something strange.

That's because that configuration entry indicates the "last changelog the user has seen". So if it says 3.6, it knows that you haven't seen the latest change log yet so it will show the news pop-up. If it's 4.5, it knows that you have already seen the change log so it will not show it to you. The 3.6 version number is *supposed* to be there sometimes. If a configuration value always needs to be a certain value (like "4.5") then we might as well not store anything.

Ghostkeeper, 2020-03-03:
I'm not able to reproduce this issue yet, apart from by manually taking away the rights of Cura to modify the configuration folders.

> When I click this menu item, I enforce writing the config, nothing else.

I don't understand. Clicking the "Show Configuration Folder..." is supposed to just open the configuration folder(s) in your file explorer. It won't do anything with writing the configuration files, right?

fieldOfView, 2020-03-03:
> I don't understand

OP is suggesting to add a button (somewhere) to force writing the current configuration to disk. What OP may not realize is that Cura already writes (attempts to write...) the configuration to disk automatically every 2 seconds (autosave) in addition to when closing Cura.

fieldOfView, 2020-03-03:
What if the presence of `cura.cfg~` confuses Cura? @azalanono, could you remove that file? It seems to be an autosave version from an attempt to edit the configuration in a text editor.

fieldOfView, 2020-03-03:
Interestingly, there is no autosaving going on in the logs posted by OP. @azalanono could you post your entire cura.cfg file?

azalanono, 2020-03-03:
I GUESS that Cura somehow gets confused when "migrating" from one appimage version 
to the next.

This might have beed initialized first time years ago (how else can a version 3.6.0 be inside 
a 4.5.0 config?) but with my "trick" of saving and recovering I somehow overruled this nasty 
popup?

An easy solution will be to have a checkbox like "Don't show this again" in the popup?

But in general... Cura quietly accepts differing version numbers inside a config without any 
warning?

Am Dienstag, 3. März 2020, 11:26:33 CET schrieb Ghostkeeper:
> I'm not able to reproduce this issue yet, apart from by manually taking away
> the rights of Cura to modify the configuration folders.
>
> I don't understand. Clicking the "Show Configuration Folder..." [...]

azalanono, 2020-03-03:
Am Dienstag, 3. März 2020, 11:45:02 CET schrieb Aldo Hoeben:
> What if the presence of `cura.cfg~` confuses Cura? @azalanono, could you
> remove that file? It seems to be an autosave version from an attempt to
> edit the configuration in a text editor.

I use 'joe' as text editor.  Joe keeps the last version as <filename>~

I have removed that file manually now, but how can a different filename 
'cura.cfg' is not 'cura.cfg~' confuse cura at all?

Anyway, I deleted ancient version directories (3.*) from my disk now and in v4.5 
I have now correct version numbers in cura.cfg.  I am interested in what will 
happen in version 4.6.

Ghostkeeper, 2020-03-03:
I don't think `cura.cfg~` would confuse Cura. Since the extension is different it should not even attempt to perform a version upgrade on it. I tried making a copy of `cura.cfg` renamed to `cura.cfg~` and everything seems to work fine still.

A checkbox like "don't show this again" also won't be able to save the user's preferences if this preferences file won't be saved.

closed 2020-06-19 by no-response[bot]