Ultimaker/Cura #7176: Print a rectangle Tower / Multicolor

Pomaroli, 2020-02-28:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])
There is no specific problem by now as more a feature request were it would be nice to can switch from a round tower while printing 2 Color to a rectangle one.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)
In the dual extruder settings it should be possible to set the shape of the tower like round, rectangle 

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)
there is no alternativ right no it prints the tower always with a round shape

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

Users with dual extrusion printer. Creality CRX

**Additional context**
(Add any other context or screenshots about the feature request here.)

fieldOfView, 2020-02-28:
You forgot to mention *why* you want this. What problem would it solve?

Pomaroli, 2020-02-28:
Depending on the Diameter of the Circle the inner circle pattern extrude's very poor. With an rectangle shape and an diagonal pattern the extrusion would be much better.
Thanks

mahtDFR, 2020-02-28:
Prime towers used to be square until some versions ago, when it was decided to replace them with circular ones for better reliability. Circular ones use gentle curves instead of straight lines, which causes less printhead jerk. Maybe you have some examples of poor extrusion? Could you not reduce the diameter?

Pomaroli, 2020-02-28:
[image]

Ye3s i tryes some different things, please see attached some pictures

mahtDFR, 2020-03-02:
Given that we made the decision a while back to switch from square to circular, it's unlikely we'll take this further.

If you feel this is a useful feature for Creality users, you can make a pull request. Please note that we won't be able to provide any quality control for this as we don't have the time or resources to test it, so it will be up to the community to test and maintain it. Alternatively you could [make a plugin](https://github.com/Ultimaker/Cura/wiki/Plugin-Directory).

Pomaroli, 2020-03-02:
No problem i can test it, can you move this request to pull request ?

Ghostkeeper, 2020-03-02:
You can test this by going back to Cura 4.1, which was the last version that had the option for a square prime tower. You can test that and see if it's any better.

The reason why you're seeing "bad adhesion" for the inner shell of the prime tower is not because it's a smaller radius or anything. It's because these inner perimeters are purged only once every 2 layers, rather than every layer. So it's not that the flow of material is bad; it's just that the material can drop down further than normal and so your line width is approximately half for the same flow.

This is not really a problem as far as I can see. Since the outer shell still always prints every layer, the stability of the prime tower is not a concern. It's really just printing a large chimney with the outer shell, then purging inside the chimney with the other extruder. It just happens to purge along the sides of the tower since that reduces the chance of blobs sticking up and out of the tower.

We can't move a request to "pull request" since a "pull request" means that you request us to pull in your code. We'd need you to write the code first then. However even if you program this out I don't think we'll merge it unless we are convinced that it's an actual improvement.

closed 2020-03-02 by Ghostkeeper