Ultimaker/Cura #7177: Cura 4.5 (unusable) makes 4.4.1 (too) unusable

BoHoehne, 2020-02-28:
**Application version**
(The version of the application this issue occurs with.)
4.4.1. & 4.5.
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 7
**Printer**
(Which printer was selected in Cura?)
Tevo Tornado
**Reproduction steps**
1. (Something you did.)
Install Cura 4.5.
2. (Something you did next.)
Tried to use Cura 4.5. (did not work).
On 4.5: It is impossible to click any menu. (No reaction)
On 4.4.1: Dito

Both applications are unusable !!

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)
Working application.
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Did anyone consider testing the application --> in advance <-- ???

fieldOfView, 2020-02-28:
> Did anyone consider testing the application --> in advance <-- ???

Yes, but not on Windows 7. Did you consider that Windows 7 is no longer supported?

BoHoehne, 2020-02-28:
How about placing an installation warning ????

BoHoehne, 2020-02-28:
The trouble started with with upgrade popup, which made Cura 4.4.1 unusable, because it locked up the entire application. Meaning that clicking on dropdown menus, profile, prepare/preview/monitor is not resulting in reactions anymore. 

Upon installing 4.5. the same happened with the newly installed application.

BoHoehne, 2020-02-28:
Cura appears to be fine. 
It seems the installer is replacing a component, that is requiring a reboot.

So, what is missing, is a --> **mandatory** windows reboot <-- upon installation !

mahtDFR, 2020-02-28:
>Did anyone consider testing the application --> in advance <-- ???

We have 2 QA testers and ran a two week long public beta, so yes.

>So, what is missing, is a --> mandatory windows reboot <-- upon installation !

A mandatory reboot isn't necessary from my experience (Win10x64). Windows 7 isn't supported by Microsoft anymore, so it's unlikely we will actively support it either.

BoHoehne, 2020-02-29:
So, the installed basis is irrelevant to you.

Everything must be in order then.

closed 2020-02-29 by BoHoehne