Ultimaker/Cura #7180: Adaptive walls ?  (Find no better naming)

azalanono, 2020-02-28:
Hello.
When printing very strong parts, I print 5-8 wall lines side by side. This is ok, but there is one huge disadvantage:
This 8 line are also visible on top and bottom layer, surrounding the top/bottom layer.

It would give *much nicer* results if cura would print upper/lower edges like this:
||||___
|||____
||_____
|______

So in 1st layer there is only one wall line '|', in 2nd layer will be two wall lines, in 3rd layer will be three, .... until the set number of wall lines or bottom layers is reached.

As result you see very homogen surfaces, will for sure look lots better.

Lissanro, 2020-02-29:
I think number of edges only in top/bottom solid layers needs to be affected, for example set to 1 or 2 walls, whatever user specifies. Anything below/above will not be visible, and can have normal number of walls. Perhaps it would be easier to implement this way.

azalanono, 2020-02-29:
Am Samstag, 29. Februar 2020, 05:53:46 CET schrieb Lissanro:
> I think number of edges only in top/bottom solid layers needs to 
be
> affected, for example set to 1 or 2 walls, whatever user specifies.
> Anything below/above will not be visible, and can have normal 
number of
> walls. Perhaps it would be easier to implement this way.

You are right.

Ghostkeeper, 2020-03-03:
What do you think should happen if only part of the shape is top surface? For instance, if you have a staircase like this:

[image]

On the highest layer shown, some parts of the layer would need to have 1 wall, while other parts of the layer need to have multiple walls. You'd either need to interrupt the wall, or make it follow along the edge of the skin through the inside. Both are a bit problematic in my opinion, for visual quality or productivity.

azalanono, 2020-03-03:
Am Dienstag, 3. März 2020, 11:35:30 CET schrieb Ghostkeeper:
> What do you think should happen if only part of the shape is top surface?
> For instance, if you have a staircase like this:

> On the highest layer shown, some parts of the layer would need to have 1
> wall, while other parts of the layer need to have multiple walls. You'd
> either need to interrupt the wall, or make it follow [...]

I would do it in a n-pass way.
     1. Compare current layer with layer+1 and generate an "xor" to get only those areas that 
will "vanish" in layer+1.
     2. Treat this areas individually with a temporary wall number setting of 1. This way the 
"filling mechanism" will fill to this one wall.
     3. Treat the remaining parts of the layer as usual. Means, give them the configured 
number of walls.

I think such method should be practical and will use most of existing code.

It would even be better to extend the above to more than "layer+1" to get at least two 
layers with only one wall line. To be sure that nothing stamps through from below. Best will 
be - my GUESS - to look ahead for all top layers. If you have 6 top layers, all 6 could be 
taken. So the fill will grow towards the physical outer edge and "tack together" the previous 
walls (will possibly increase stability).

Ghostkeeper, 2020-03-04:
So something like this:
[image]
(Achieved with a cutting mesh.)

The problem I want to highlight is in step #3 of your description. The remaining parts of the layer have more walls. Those walls have to start and end somewhere, or they need to become loops. Starting and ending is problematic for getting gaps in your walls, because the line typically gets dragged along by half a line width or just fails to extrude immediately after an unretract (depending on the material). Looping is what's shown in the image. This introduces a seam where the two sets of walls are split up. This is less problematic for print quality since the seam would only be for the top 3 layers or so (depending on the number of walls). However then let's say you get a model with a curved top side like this:
[image]
Then you'll get extra walls on the inside everywhere which would take a LONG time to print.

azalanono, 2020-03-04:
I cannot really figure out the problem.   Nothing "suddenly appears", everything grows from bottom to top, either based on support or previous layers.
So we have at least one outside wall (surface of the part), and the part has a preset number of wall lines, i.e. 6 lines for a strong part.

When an area is 100% filled in one layer, cura currently draws all 6 wall lines, sometimes even adds an additional line to surround the area to be filled, and finally uses some algorithm to fill the area.

Nothing has to change, except of this:
The **visible part of filled areas** has to be extended towards the outermost wall line. So in this region the inner wall lines will be replaced by a solid fill to the outermost wall line.
I see no stability issues here, the amount of material is identical.

printingotb, 2020-03-05:
This can already be done and it's quite simple.
If I understand your question correctly, you ant to use many walls but you don't want them to be visible on the topmost layer?

-Set number of wall lines to one or two. 
-Enable infill walls and replace the missing wall lines with them. If you don't want it to print normal infill, set the pattern to concentric at 0.1%. This removes infill completely but keeps the infill walls.

Bonus tip, increase infill wall line by around 50% and reduce the number of overall walls until you get the same total thickness. You will print equally as strong parts in less time.
When I discovered this, I was able to reduce the print time by one third while getting nicer surfaces (slowed down outer wall) and a part that was also slightly stronger than before (increased overall wall thickness by a small amount extra).

You may need to increase temp to compensate for the larger amount of material being extruded in the same time. Or slow down some but you will still get the part printed faster.

"sometimes even adds an additional line to surround the area"
This is the extra skin wall count, you can set it to 0 to remove it.