Ultimaker/Cura #7183: Moving Cura 4.5 to a different screen - Crash an no workaround.

DX12Diodes, 2020-02-28:
**Application version**
Cura 4.5

**Platform**
Windows 10
i7 CPU 16GB Ram Nvidia 1070GPU

**Printer**
Ender5

**Reproduction steps**
1. Open Cura
2. Move it to another screen
3. Close Cura.
4. Try to open it again. It tries to open it on the screen you moved it too but crashes and never loads. Just a white screen.
This happened in a previous version (i raised a ticket then too) and you had to uninstall Cura completely including config and you could get it to work. Then you added the tick box not to remember screen config.
In this version however there is no option to remove Cura and all the config, it just uninstalls. When you reinstall it remembers the screen it was on so still crashes, i'm now stuck in the loop and cant get out? Is there a reg key somewhere i can delete to start again? 
Next time ill remember to tick the box not to remember screen settings! but this is an issues that needs fixing. 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
As above in steps.

**Expected results**
The screen the program is on should be irrelivent is should work on any screen.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-02-28:
There is a workaround:
In the General preferences, uncheck the option "Restore window position on start"

fieldOfView, 2020-02-28:
You can manually set that option by opening `%appdata%\cura\4.5\cura.cfg` in a texteditor, and adding the following line to the `[general]` section:

`restore_window_geometry = False`

DX12Diodes, 2020-02-28:
As i said in the original post. I cant get into Cura to select that option because its already at the stage where its crashing. 
Uninstalling Cura and reinstalling in this version doesnt reset the screen settings so after a reinstall the problem persists. So i can never get into Cura to select the option.

DX12Diodes, 2020-02-28:
> You can manually set that option by opening `%appdata%\cura\4.5\cura.cfg` in a texteditor, and adding the following line to the `[general]` section:
> 
> `restore_window_geometry = False`

That is what i needed! you tha man thank you!

Ghostkeeper, 2020-03-03:
We have yet been unable to reproduce - and consequently fix - the issue with multiple graphics cards assigned to each screen that would supposedly crash Cura.

It seems you're secretly trying to report a different bug here though: Doesn't the uninstaller provide you with an option to erase the settings?

fieldOfView, 2020-03-03:
> Doesn't the uninstaller provide you with an option to erase the settings?

Does the one provided by the MSI?

Ghostkeeper, 2020-03-09:
Ah, no indeed it doesn't apparently.

RevBooyah, 2020-03-10:
I can repeatedly get the same issue on both 4.4.1 and 4.5.  Though, I didn't need to uninstall - only shut off the other screen, so it would load on the primary monitor.

ninovanhooff, 2020-03-11:
Developers: for the uninstaller leaving config files issue; I made internal ticket https://jira.ultimaker.com/browse/CURA-7293

scottjmckee, 2020-05-13:
I'd like to contribute my voice to this issue. I have multiple monitors on a single GPU (GTX 1080) and if Cura starts on anything but the primary monitor it will fail as described above. My fix was to disable all other monitors to force Cura to open on the primary.

fieldOfView, 2020-05-13:
@scottjmckee have you tried unchecking the "Restore window position on start" checkbox in the general preferences? As mentioned above, you can manually set that option by opening `%appdata%\cura\4.6\cura.cfg` in a texteditor, and adding the following line to the `[general]` section:

`restore_window_geometry = False`

scottjmckee, 2020-05-13:
I have done that, I wanted to reply to this issue to add another voice to the issue of Cura having problems with multi-monitor setups. I often like to have my Cura on a side monitor to watch while I continue design work on my projects.

fieldOfView, 2020-05-13:
With this workaround, you can still drag Cura to your side monitor once it has started. It will just start on the primary display.

Ghostkeeper, 2020-05-14:
May be fixed by upgrading to Qt 5.12: https://bugreports.qt.io/browse/QTBUG-50371

ameerabdallah, 2020-08-11:
I found a solution to this issue. Turns out it has almost nothing to do with cura and everything to do with other applications you have installed on your computer. I know how to recreate the issue now too. If you have the Asus Sonic Suite, Realtek Audio Drivers or Nahimic Audio Software you will get crashes whenever you open a window on the second monitor. I had seen this solution [here](https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/Fusion-360-user-interface-menus-don-t-work-on-a-Windows-MSI-GT72S-Computer.html) where it was affecting fusion360 as well. Here are the steps to fix this

1) Open Control Panel
2) Open Programs and Features
3) Look for anything involving nahimic audio software, asus sonic suite, or realtek audio drivers
4) Uninstall all of these
5) Either restart your computer or go into task manager and look for any process that has nahimic in its name and end the task

This should fix your issue. There may be other audio softwares that use nahimic audio software so really you can go ahead and remove all audio software and then reinstall one by one to see which ones are causing the crashes or just deal with not using the audio software, audio quality really isnt gonna be affected by it anyway

Ghostkeeper, 2021-06-16:
We've included a possible fix for this issue in Cura 4.10. We don't know if it really fixes this issue, but there is hope. The beta should be released soonish so you can try for yourself then.

scottjmckee, 2021-06-16:
Thank you for the heads up :)

Ghostkeeper, 2021-06-18:
The 4.10 beta is now released, if you'd like to try.

scottjmckee, 2021-06-18:
I just grabbed the beta and tried opening and closing from various
positions on my monitors. It looks like the issue is now resolved so thank
you :)

closed 2021-06-21 by Ghostkeeper

usernametakenaaaa, 2022-05-28:
Looks like they re-introduced that bug in Cura 5
As soon as Cura 5 defaults to opening on my second monitor it hard hangs until you kill the process
Main Monitor 1 is a Nvidia card, Monitor 2 is onboard intel

The main crash can be avoided by opening the config file and removing the "left" setting of the window.
However, it will crash anyway when it opens any other dialog there which you can not control. Some dialogs open left side (all good) a few open right side (cura crashes)