Ultimaker/Cura #7184: Unnecessarily long moves

azalanono, 2020-02-28:
Hello.
When printing multiple parts side by side, I almost always watch a very strange behavior when the print head moves from one part to the next.
In both parts printing starts close to front-right edge, part #2 is right of part #1, only to give you an idea of arrangement.

When the layer of part 2 is finished, the print head follows part 2's walls to back-left edge, then it surrounds(!) half of part 1, and finally follows part 1's wall to the start point of next layer. >8O
This is traveling >100 mm in bizarre path instead of simply moving along the front wall towards part left and there (again along the wall) to the starting point. This would be less than 20 mm.

Why is this happening? Can't such moves also be optimized? The actual move needs 5-6 times longer as necessary.

smartavionics, 2020-02-28:
Please save that project file (File -> save) and then zip up the .3mf file or rename to .zip and attach to this issue so we can understand what you are saying. Thanks.

sgtnoodle, 2020-05-12:
I noticed some rather silly travels while printing the first layer of this print. It's avoiding travelling over features that haven't been printed yet. Perhaps it's the same bug?

[image]

closed 2020-06-19 by no-response[bot]