Ultimaker/Cura #7185: Alterating moves in every even/odd layer?

azalanono, 2020-02-28:
I have the problem that first millimeters of extrusion don't have the full amount of material (until the pressure has increased again inside the nozzle).
Sadly Cura tends to do always the same moves in succeeding layers, so the printed part gets a real problem when the walls are very thin (also has effect on infill and support lines, they are only single stroke.

It would for sure improve printing quality if ...

- Cura would use different preferred printing directions in even/odd layers. I.e. a wall is printed CCW in layers 1,3,5,.... and CW in layers 2,4,6,....  This way the reduced material will not always be on same spot.

- Another idea will be some sort of rotating "gravity". N, NE, E, SE, S, SW, W, NW   to talk in cardinal points and to start all printing from there towards the opposite direction. (can be good for top/bottom layers, Infill/support printing, but sometimes also for printing walls.

Ghostkeeper, 2020-03-03:
For the walls you can fix this by setting the Z Seam Alignment to "Random". While designed to spread around the seam so that it's less visible to the human eye, it will also have a positive effect on strength if that's what you're asking.

For the infill you can enable Randomize Infill Start. This is more or less exactly what you're asking. It's designed to eliminate the weak point in the infill.

Another think you can try that directly attacks your problem is to increase Retraction Extra Prime Amount. After every unretract it will extrude slightly more material to compensate for the material that was oozed out. You might also reduce the speed of unretraction.

Alternating the direction of rotation can improve this or not, depending on the direction of the preceding travel move which is arbitrary. It may also worsen it. I don't think that's really a solution to this problem.

smartavionics, 2020-03-03:
Wherever there is un-retracted travel, what comes next will be under-extruded. So, if before the wall there is a long un-retracted travel move, the nozzle pressure will be low. If you are using combing, I recommend setting the max comb distance without retraction to quite a low value, i.e. 10. Setting retract before outer wall can also be beneficial. For infill, I recommend using the connect infill lines option to avoid un-retracted travel.

loneFunction, 2020-03-04:
I have been battling this issue as well and managed to track a few stray checkboxes that were enabled at some point, but were not understood at the time.
The first that gave the most headaches was Flow Rate Compensation in the Experimental section. For some reason a large number was put into the max compensation field and it was causing major under extrusion after most travels.
After fixing that I double checked retraction settings because I was getting a lot of pits and pops after de-retractions. I went from 5 to 1mm and 50mm/s down to 15mm/s to fix that issue.

If you have a really oozy nozzle/hotend the extra retract prime can compensate, but it will add that amount to every retraction regardless of distance traveled. The nozzle wont have time to ooze during short moves, so they will be overextruded, while long moves will still be underextruded.
There is an older plugin called ScalableExtraPrime which addresses this by scaling the extra prime amount based on travel distance. I personally think the extra prime feature should be replaced by this because rarely would you ever want a dynamic thing like ooze to be controlled by a static value.

Ghostkeeper, 2020-09-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/6672.

closed 2020-09-11 by Ghostkeeper