Ultimaker/Cura #7186: Anticipating horizontal holes

FlailAway, 2020-02-28:
**Is your feature request related to a problem? Please describe.**
I have a flat project 12mm high and at 5.5mm there are several 6mm diameter horizontal holes. I have the infill set to 5% as it is a very big print and 5% will be quite rigid enough. But at layer 0.5 in the vicinity of the holes it struggles to bridge the 5% infill. See pic

**Describe the solution you'd like**
For horizontal holes it would be nice for them to be seen in advance and build a supporting base from the second layer up, or some method of support to pad out the infill in that region.
Or, seeing all holes as unique items with their own wall-count. This would also help for holes that need to be tapped after printing.

**Describe alternatives you've considered**
The Mesh option would solve it, but it must be done for every hole and after every Slice.

I have upped the wall count so that the holes get a lot of padding sagging (see pic) into the 5% so the holes get printed kinda OK. But, see the "tapped" comment above **and** then everything gets more walls. That wastes plastic where it is **not** needed and adds time.

**Affected users and/or printers**
Anyone trying to make horizontal holes

[image]

smartavionics, 2020-02-28:
I don't really understand what you are saying, is it possible to attach the project file? Thanks.

FlailAway, 2020-02-29:
Thanks, but look at the black ellipses and see how the first outer layers of a hole does not span the infill. It is sagging down to the last bottom layer as expected but not  desired. :) 

My first thought was for the slicer to "know" that a horizontal hole was coming a few layers up and would build support under from the base it so there is a flat surface to start the hole walls on when it gets to that layer where the holes are about to be made.

As it is now, with 3-Walls set for the entire print, 2 of those hole-horizontal walls have pretty-much  already fallen down to the flat surface below and the lower side of the holes effectively has no wall on the lower section. It comes out having only about 300-degrees of wall on the upper section of the entire hole. It is OK at the outer wall as that section is supported by the outer wall, much like the Support I am suggesting.

It would be handy for holes to have their own settings as in wall-count. If I want to tap holes after printing I have to use 5-walls for the entire print so there is enough "meat" in holes beyond the cut thread. If we had "Wall count = 2" and then "Hole-wall-count = 5" everything would be printed with 2-walls except holes which would have 5.

smartavionics, 2020-02-29:
Please provide a project file so I can slice it and get an understanding of the 3d structure. If a picture is worth 1000 words, a project file must be worth 1e6 (at least).

FlailAway, 2020-02-29:
I am ahead of you. I was just creating a small sample of the issue. In the Previww, look at layer-8. There is a nice flat surface there for the start of the hole walls. But in reality, that flat surface and the next on up will both sag into the openings of the Infill.

Hadta rename it .log as this git doesn't like .obj.

[test.log](https://github.com/Ultimaker/Cura/files/4270490/test.log)

smartavionics, 2020-02-29:
I got the object but I don't know if the scale is right and what layer height to use, etc.

My Cura builds for some time now where walls are added to infill where there is skin on the layer above. I believe that feature has now made it into Cura 4.5.0. Here is what it does on the layer below the first skin layer of the blind hole.

[image]

Search for skin edge support in the settings.

FlailAway, 2020-02-29:
The holes are horizontal, please check the pic in the OP. That looks to be sliced vertically. I am using 0.4 nozzle, 0.2 layer, 2-walls, 3-top and bottom and Lines for infill at 10%.

My printer is out in the garage 300-feet away with no internet connection and no heating. Currently 11-F outside so trips out there are not well received.  :)

smartavionics, 2020-02-29:
I just sliced it how it was loaded.

Anyway, the skin edge support is beneficial when the part is horizontal, this is the layer before the first skins...

[image]

smartavionics, 2020-02-29:
And here's the first of the skin layers above...

[image]

And a few layers later come the first layer of walls for the holes...

[image]

No droopy skins even with fairly low density infill.

FlailAway, 2020-02-29:
Are you using the stock 4.5.0? I do not get that first layer outline, "this is the layer before the first skins".

> I just sliced it how it was loaded.

I realized it was vertical when I loaded it into Cura but it was so damned cold out there I didn't bother going back to FreeCAD and redraw it at the correct orientation. I figured the use of the words "horizontal hole" would enthuse you to first lay it down before slicing.

This seems a waste of time, it is not what your version is getting, it is what is happening here. The pic in the OP is clear enough that sagging is happening. It is also not what the Preview is showing, it is what is actually happening when printing. The Preview is faked based on layer line data and **not** molten PLA.

I quit.

closed 2020-02-29 by FlailAway

smartavionics, 2020-02-29:
These are the skin support settings I am using...

[image]

Ghostkeeper, 2020-03-03:
There are two ways in which you can prevent this. One is the "Skin Edge Support Thickness" setting that Smartavionics has already highlighted. It's not entirely what you're asking as you were simply asking about supporting the entire surface where the skin is going to rest on, but I think in your case with the slim hole it will work well. This feature is indeed new in Cura 4.5.

The other method is to increase the Gradual Infill Steps. This will halve the infill density at intervals shortly below the skin. So for instance, set Gradual Infill Steps to 2, then multiply the infill density by 4. This should give you a denser infill just underneath the horizontal hole to support the skin there, but for the rest of the model it will have the same low infill density as you are currently using. Gradual Infill will also support the top side of your model the same as this hole though; to a computer those cases are the same, it just needs to support the top surface.

Similar to Gradual Infill you can also use the Cubic Subdivision infill pattern at higher infill densities. This then uses lower infill densities more towards the middle. The main difference is that it will increase the density on all sides rather than just the top surface so that's not quite what you need now.