Ultimaker/Cura #7188: [4.5.0] Error on Install

acedra, 2020-02-28:
**Application version**
4.5.0

**Platform**
Win10 64bit
Intel i7-8750h @ 2.2GHz
16GB ram

**Printer**
none yet but will be Ender 3 Pro

**Reproduction steps**
1. Download and launch Ultimaker_Cura-4.5.0-win64.msi
2. Agree and start install
3. Computes space and then errors with "Setup ended prematurely. Please try again."

**Screenshot(s)**
none yet

**Actual results**
Click on and it exits.

**Expected results**
Cura should install.

**Project file**
none

**Log file**
no log file that I know of since it won't install.

**Additional information**
I did not uninstall 4.4 before trying to install 4.5. I have kept several previous versions installed in the past and this has never been an issue.

fieldOfView, 2020-02-29:
I think this is the first time we have 1,000 issues open:
[image]

WHWFD, 2020-02-29:
I'm having a similar issue.
I've had 4.3.0 and 4.4.1 installed. Both of them were ok, but suddenly closed two days ago, after showing only a white screen after loading the settings.
Switching to 4.5.0 was no real solution, I'm in the same situation here. Repair or uninstall didn't work.
For a short time I could bring it further up by starting with another Windows user profile, but now it's all the same:
Loading - Some setups - Blank screen - Gone.
What can be done? Thanks for hints

smartavionics, 2020-02-29:
Hello @WHWFD , can you please locate stderr.log and cura.log that should be in AppData/Roaming/Cura and AppData/Roaming/Cura/4.5.0 respectively. Zip them up and attach to this issue. Thanks.

WHWFD, 2020-02-29:
Both Logs in attachment.
Thanks for immediate reply.
[4.5 Logs.zip](https://github.com/Ultimaker/Cura/files/4270262/4.5.Logs.zip)

smartavionics, 2020-02-29:
Thanks for the files. I can see where it blows up but, unfortunately, it doesn't provide enough info (to me, at least) to identify the cause of the problem. Sorry.

You could try installing one of my builds which are built differently from the UM releases and they may provide a workaround. They can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

Ghostkeeper, 2020-03-04:
> I've had 4.3.0 and 4.4.1 installed. Both of them were ok, but suddenly closed two days ago, after showing only a white screen after loading the settings.

This issue is different from the original. Please refer to https://github.com/Ultimaker/Cura/issues/5296 for more information.

Ghostkeeper, 2020-03-04:
Rather than the .msi you can also use the .exe installer that we provide with the releases. The .exe works the same way as it has always done. We basically just built the .msi installer in order for system administrators to be able to install Cura automatically for their users in the network.

We're unable to reproduce this error so far. Our testers install Cura multiple times per day on different computers.

WHWFD, 2020-03-04:
It's better now. Cura 4.5.0 has come back again, 4.3.0 and 4.4.1 also.
I couldn't test much in the recent days, and it looks like I'll have to configure the Start GCode for my CTC Bizer with Sailfish firmware.
Thanks for the hints and the link to #5296 
Walter

acedra, 2020-03-08:
> 
> 
> It's better now. Cura 4.5.0 has come back again, 4.3.0 and 4.4.1 also.
> I couldn't test much in the recent days, and it looks like I'll have to configure the Start GCode for my CTC Bizer with Sailfish firmware.
> Thanks for the hints and the link to #5296
> Walter

Thanks for hijacking my thread. Next time start your own maybe?

acedra, 2020-03-08:
> Rather than the .msi you can also use the .exe installer that we provide with the releases. The .exe works the same way as it has always done. We basically just built the .msi installer in order for [...]

I'm sorry, can you tell me where the .exe installer is located? I looked on the website and was unable to located anything other than the .msi download. Thanks!

WHWFD, 2020-03-08:
Hi Mark,

I Think, I’ve found the cause for Cura stopping here. There seems to be an interaction with GPU Tweak II (ASUS). Read more here:

https://community.ultimaker.com/topic/27929-cura-friert-nach-aufstart-direkt-ein/?tab=comments#comment-240042

 

By the way: Do you know a solution for Cura operating with 3Dconnexion devices? I’m here with a Spacemouse Enterprise. The menu topline there shows Cura, some buttons are active, but no knob response.

 

Have a nice day

Walter

 

Dipl.-Ing. Walter Holzwarth
Ingenieurbüro für Maschinenbau
Brombeerweg 23
D-37281 Wanfried
Tel. 05655 92144 

Mobil 0163 9641823

http://www.holzwarth-cad.net <http://www.holzwarth-cad.net/>

smartavionics, 2020-03-09:
> By the way: Do you know a solution for Cura operating with 3Dconnexion devices? I’m here with a Spacemouse Enterprise. The menu topline there shows Cura, some buttons are active, but no knob response.

See https://github.com/smartavionics/RawMouse

WHWFD, 2020-03-09:
Good morning Mark,

meanwhile I tried it, but no change. I’m not sure about placing the files in the right foders.

 

Thanks

Walter

smartavionics, 2020-03-09:
> meanwhile I tried it, but no change. I’m not sure about placing the files in the right foders.

We shouldn't discuss the mouse here in this issue, please open an issue at https://github.com/smartavionics/RawMouse and provide the cura log file so I can check it's finding the mouse OK.

Ghostkeeper, 2020-03-09:
> I'm sorry, can you tell me where the .exe installer is located? I looked on the website and was unable to located anything other than the .msi download. Thanks!

You can find the normal installer here: https://github.com/Ultimaker/Cura/releases/tag/4.5.0

I'm looking into why this installer is no longer provided on the Ultimaker website. In my opinion this installer should be the default, and if a system administrator needs silent installation so badly he can find it on Github...

mahtDFR, 2020-03-09:
The MSI we hosted on [Ultimaker.com](https://ultimaker.com/software/ultimaker-cura) has caused some confusion and issues, so I've just swapped it out for the exe.

People who want the MSI can go to [Github](https://github.com/Ultimaker/Cura/releases/tag/4.5.0) and find it.

mahtDFR, 2020-03-10:
It's not reproducible here so far and there's not a lot of data to work with. Maybe someone who has experienced the issue can help by sharing MSI logs with us. The MSI doesn't run with logging by default, but you can enable it running via the command line in Windows:

`<MSIdirectory> /lxv <OutputDirectory>\cura45msilog.txt`

Where `<MSIdirectory>` is the directory your MSI is stored and `<OutputDirectory>` is the output directory.

e.g. If the MSI is on my desktop, I ran this in the commandline `Ultimaker_Cura-4.5.0-win64.msi /lxv "C:\Users\m.jani\OneDrive - Ultimaker B.V\Desktop\msilog.txt` Your path will vary.

If you can share this when the issue occurs, it will help us a lot!

acedra, 2020-03-10:
> You can find the normal installer here: https://github.com/Ultimaker/Cura/releases/tag/4.5.0
>
> I'm looking into why this installer is no longer provided on the Ultimaker website. In my opinion this [...]

Thank you so much!!! This solved the problem. Got it installed and working.

Ghostkeeper, 2020-03-11:
That solved it for you, but the MSI then still fails to install I suppose.

acedra, 2020-03-11:
That is correct. So, probably something with my machine this time, since
all the others have worked flawlessly and the .exe worked. If it happens on
the next release I will update.

 - Dan

closed 2020-06-19 by no-response[bot]