Ultimaker/Cura #7189: Z homing issue on IDEX or tool changing machines

mtw3d, 2020-02-28:
**Application version**
4.4.1

**Platform**
Windows

**Printer**
Custom FFF with IDEX

**Reproduction steps**
Select any settings that results in T1 being the first extruder used.

**Screenshot(s)**
Not relevant

**Actual results**
Cura inserts the first active extruder as the very first line of the gcode. So if your settings result in T1 being the first used, T1 is inserted as the very first line in the gcode. 

However if you have an IDEX or tool changing printer that uses a probe on T0, the printer than tries to home the Z axis using T1, with obviously (and potentially dangerously) unsuccessful results.

**Expected results**
The start gcode should be the very first thing executed. Any automatically generated code should be added AFTER the start gcode, otherwise you can be overriding the users intention.

I attempted adding:

```
T0
G28
{initial_extruder_nr}
```

to my start gcode, but that just resulted in the number "0" being added to the resulting gcode file

**Project file**
Not relevant

**Log file**
Not relevant

**Additional information**
Not relevant

Ghostkeeper, 2020-03-04:
> I attempted adding:
> 
> ```
> T0
> G28
> {initial_extruder_nr}
> ```
> 
> to my start gcode, but that just resulted in the number "0" being added to the resulting gcode file

In Cura 4.5 we made some fixes to `{initial_extruder_nr}` to make it more accurate if not all extruders are being used in the final print result (as opposed to just which extruders are enabled/disabled). Can you try it there?

mtw3d, 2020-03-04:
That worked. 

In case anyone else runs into this, here is the exact start gcode required for homing:

```
T0
G28
T{initial_extruder_nr}
```

closed 2020-03-04 by mtw3d