Ultimaker/Cura #7191: Supports floating in the air.

alkhimey, 2020-02-29:
**Application version**
4.5

**Platform**
Ubuntu

**Printer**
Creality-Ender-3, 0.4mm nozzle, generic PLA.

**Reproduction steps**
1. Download the following model: https://www.thingiverse.com/thing:929504
2. Slice with supports.

**Screenshot(s)**
[image]

**Actual results**
Supports are floating in the air. Can't print like this.

**Expected results**
There is a connection between the supports and the bed.

**Project file**
(Change extension to 3mf)
[CE3_fidget_star_standing_30_2_50.zip](https://github.com/Ultimaker/Cura/files/4270299/CE3_fidget_star_standing_30_2_50.zip)

mahtDFR, 2020-03-02:
Thanks for the report. I was able to reproduce it. 
We thought it might be a result of stair stepping issues (see 6379 for more discussion on this). Even when stair stepping was set to 0 the issue persisted.

[image]

BagelOrb, 2020-03-03:
This is due to the Support XY distance.

It would be nice if Cura checked for something like this happening and always generates support which is supported, but that's quite a difficult thing to implement.

Ghostkeeper, 2020-03-05:
Workaround for now is to use Tree Supports or Enable Conical Supports with a slight negative conical support angle. Solving the issue where the support is removed by a protrusion in the X/Y plane is a pretty hairy problem that Tree Support has a solution for.

I expect that in practice, this support will probably repair itself at some point though. It will leave a slight scar.

alkhimey, 2020-03-06:
@Ghostkeeper It seems that the conical option prints the model engulfed inside a conical support structure. Are you sure it is possible to remove the supports afterwards?

BagelOrb, 2020-03-07:
I think you should lower the angl setting of the conical or tree support then.

Ghostkeeper, 2020-03-09:
I think you should increase the normal Support Overhang Angle since I'm not seeing the angles at the top being supported if I only add conical support to your previously submitted project file:
[image]

Though in actuality if you really want to print this object in this orientation, you'll probably need to print it like this or it will become very unstable towards the top.

StarNamer, 2020-04-14:
I'm not sure this is that same issue, but I've had an instance of Tree Supports being generated floating in mid-air in 4.6 Beta. So the recommendation to use Tree Supports to avoid the issue won't necessarily fix it.

Here's a couple of images of Cura showing the floating support, plus a photo taken while it was being printed with the c(almost) corresponding view in CURA. Fortunanlely, the print succeeded anyway; unfortunately, re-slicing did not produce the same support structure - presumably because tree supports have some randomness - so I can't provide a model which reproduces the issue.

[image]
[image]
[image]
[image]

StarNamer, 2020-05-14:
It seems that Tree Support is often "optimistic" about print supports in mid-air.
[image]
[image]

Ghostkeeper, 2020-05-18:
The second and third images, no there isn't. Usually it corrects itself fairly quickly. The issue is unrelated to this issue though. Please refer to [this issue](https://github.com/Ultimaker/Cura/issues/4342).

spiritedfang, 2020-07-02:
I found the same issue in version 4.5.0. It leaves floating support structures in every support type under the head and wings in this model at 200% scale for the Ender 3: https://www.thingiverse.com/thing:3931684

alkhimey, 2020-07-04:
Yep, I have this issue again in 4.6.1.
I am not sure, but it looks like the conical supports experimental option is not doing anything, the supports look suspiciously similar to the zig zag supports:
[image]
This one without conical box ticked:
[image]

Project:
[CE3_Lejon.3mf.zip](https://github.com/Ultimaker/Cura/files/4872238/CE3_Lejon.3mf.zip)

BagelOrb, 2020-07-04:
That's a wholly unrelated issue. Could you please make a separate ticket for that?

CiriousJoker, 2020-12-26:
[image]

Using Cura 4.8 on Big Sur.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

StarNamer, 2024-11-22:
I haven't noticed this in recent versions so I'd not object to closing it.

closed 2024-11-23 by GregValiant