Ultimaker/Cura #7192: Add reload dialog for AMF models

jcharaoui, 2020-02-29:
When overwriting an STL model file which is open in Cura, Cura detects this change and displays a helpful dialog offering to reload the model on the build plate. This doesn't work for models loaded from AMF files. Cura doesn't detect when the AMF source file changes.

It would be nice for Cura to also detect when the AMF source file is changed, and offer to reload the model on the build plate, just like with STL models.

fieldOfView, 2020-03-01:
Good catch. See PR #7205 (which also fixes DAE and the other formats supported by Trimesh)

Ghostkeeper, 2020-03-05:
We got this reported internally a month ago as well, but it ended up on our backlog since then. Thanks for the fix, @fieldOfView. I'll give that a test.

commit 7cab2b87 (2020-03-05): Fix watching for changes in AMF and trimesh-loaded files

Fixes #7192
PR #7205 (2020-03-05): Fix watching for changes in AMF and trimesh-loaded files
This PR fixes the AMF and TriMesh readers to support watching the filesystem for modified files.

Fixes #7192
closed 2020-03-05 by Ghostkeeper