Ultimaker/Cura #7194: Cura 4.5 generated gcode file finishes everytime on layer 23

SchlSE, 2020-02-29:
**Application version**
Cura 4.5

**Platform**
Win 10 6700k GTX1080

**Printer**
Ender 3

**Reproduction steps**
1. load object
2. sliced it
3. uploaded it on the Duet2 Board
4. started the print

**Actual results**
at layer 23 WDC showes 100% and "23 from x lines" and " y mm of z mm"
depending how tall the model is, you can see the infill
x > 23 ,  z > y =>  there should be more lines to be printed and the amount of material wasn't exceeded.

**Expected results**
the print should go beyond 23 line until it is really finished

**Project file**
[https://send.firefox.com/download/eaad8daf48c7602e/#GtoTVKBquo7H537-dE0q3A](url)

**Additional information**
The Reprap firmware wasn't updated betwenn cura versions , unwanted behaviour wasn't happening on Cura 4.4.1 and isn't on Prusa Slicer

smartavionics, 2020-02-29:
Nothing to do with the pause at height script you have in that project, I suppose?

smartavionics, 2020-02-29:
Also, looking at the gcode created by the pause at height script, it ends by setting the extruder mode back to absolute which is going to screw up because you are using relative extrusion.

SchlSE, 2020-02-29:
"Nothing to do with the pause at height script you have in that project, I suppose?" I didn't activate pause at height and it wasn't activated on cura 4.4.1 so I am more confused about this. also different new projects had this behaviour, that means pause at height is automatically activated.

"Also, looking at the gcode created by the pause at height script, it ends by setting the extruder mode back to absolute which is going to screw up because you are using relative extrusion."
 I never used the pause at height function. so no idea.

SchlSE, 2020-03-01:
ok, it was the standard pause feature, I still don't know why it was checked

closed 2020-03-01 by SchlSE