Ultimaker/Cura #7196: Brim is well spaced lines instead of a solid piece

bluehubbardsquash, 2020-03-01:
**Application version**
Cura 4.5.0

**Platform**
Windows 10 64bit. Intel HD Graphics 3000

**Printer**
Monoprice Select Mini V2

**Reproduction steps**
1. Build plate adhesion type selected: Brim
2. Brim width: 10mm
3. Brim line count: Autopopulated 29 lines
4. Brim only on outside: Selected

**Screenshot(s)**
[image]

**Actual results**
Brim is printed as spaced lines. These are easy to peel off the build plate. However, this type of brim does not provide the same degree of build plate adhesion as the type of brim that used to be printed which was a solid thin surface surrounding the object.

**Expected results**
I expected a solid brim.

**Project file**
[5Perch.Z6cmMONOPLAsp30in00.zip](https://github.com/Ultimaker/Cura/files/4271206/5Perch.Z6cmMONOPLAsp30in00.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4271202/cura.log)

**Additional information**
This type of brim does not provide enough build plate adhesion. In my experience it has led to objects detaching from the build plate on a number of occasions.

fieldOfView, 2020-03-01:
Because your layer view is in compatibility mode, it is hard to judge the width of the extrusion. Have you actually printed the file? If so, how does it look?

bluehubbardsquash, 2020-03-05:
The width of the brim is accurate. The brim itself is multiple parallel lines as seen in the image uploaded. I printed two other objects I know well and use as standards. I sliced both with Cura 450 and included a brim. They printed fine (please see uploaded second image). This perplexes me since I thought that the model doesn't influence the brim but in the Perch model it may be reasonable to think the model makes the brim print incorrectly.
Odd brim printing with the Perch model.
[image]
Accurate brim printing (Left: pre-Cura450, Middle and right: Sliced with Cura 450)
[image]

Ghostkeeper, 2020-03-06:
This also shows in the modern layer view, by loading that g-code into the Monoprice Select Mini v2: 
[image]

Maybe you have a project file for those fish, so that we can see if any of the settings are causing this?

bluehubbardsquash, 2020-03-12:
The Cura project files for the fish is attached. The original STL file from Thingiverse:
http://www.thingiverse.com/thing:1701205
[5Perch.Z6cmMONOPLAsp30in00.zip](https://github.com/Ultimaker/Cura/files/4321728/5Perch.Z6cmMONOPLAsp30in00.zip)

Ghostkeeper, 2020-03-17:
I loaded that project file and added the fish from that Thingiverse link, but it's not reproducing the problem.
[image]
I don't think these are the settings you were using though. Support and platform adhesion were disabled in your project, so I enabled platform adhesion to try to find your bug, but it's not happening. But that might indicate that you also changed other settings that are not in this project file.

You can also just save the project file while looking at the bug in Layer View. This will store all of your current settings and the models on your build plate too.

bluehubbardsquash, 2020-03-21:
I recreated a gcode file with the brim error. The project is included. I don't really work with Cura Project files. I just open the model (usually .stl) configure it how I want then slice, save, close. I didn't include any photos of the print as three times it turned into a ball of PLA but starting out, the brim was the same as the images shown above.
[5NEWPerch.Z6cmMONOPLAsp30in00.zip](https://github.com/Ultimaker/Cura/files/4363354/5NEWPerch.Z6cmMONOPLAsp30in00.zip)
Thank you for your time and patience. Let me know if you need any more information.

Ghostkeeper, 2020-03-24:
If I slice that project file it looks correct in layer view. Only when I save the g-code and load it back in do I see your problem.

Something is up with the g-code visualiser there I think. Here's how it looks from the side (after loading the g-code file back in):
[image]

The g-code appears to be floating.

For comparison, here is the same g-code when loaded in a different g-code viewer, RepetierHost:
[image]

The g-code is not actually floating. This can also be seen in the actual g-code where it only moves to Z0 to do that prime, then moves to Z0.263 just before starting the brim, which is the correct initial layer height. Calculating the E values by the layer height, line width and line length reveals that it's correct. Nothing is floating.

Why your actual print is going wrong then I don't know though. Could still be the bed leveling.

bluehubbardsquash, 2020-03-28:
Hello. I'm thinking that it shouldn't be a bed leveling issue. Other models with a brim print fine. Also, the brim does print and it does adhere to the plate, it just prints as separated lines which don't adhere to the base as well and so eventually it will detach in most cases. Please refer to the photos above showing how the brim prints.
I'm puzzled why the model would cause issues with the brim is added during slicing variable selection. Thank you for all the help.

Ghostkeeper, 2020-06-29:
This got closed automatically due to a request for more information to which you responded.

However I'm still unable to reproduce your problem with the g-code, only with the visualisation in the layer viewer. Where in the g-code is it going wrong? As far as I can see, the E values seem to be correct.

closed 2020-07-13 by no-response[bot]