Ultimaker/Cura #7197: [4.5.0] Printer connected via USB is not detected

rgov, 2020-03-01:
**Application version**
4.5.0

**Platform**
macOS 10.15.3

**Printer**
Monoprice Maker Select V2

**Reproduction steps**
I attached a mini USB cable to my printer to my laptop (via a USB-C adapter).

**Actual results**
Cura does not respond to the fact that the printer is attached, even after relaunching.

**Expected results**
Cura should realize the printer is attached.

**Log file**
I do see this:

```
<USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x600012575728>>] added.
2020-02-29 21:53:39,362 - DEBUG - [Thread-4] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started.
2020-02-29 21:53:40,990 - DEBUG - [Thread-4] USBPrinting.AutoDetectBaudJob.run [50]: Checking /dev/cu.usbserial-AH05LL5O if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
2020-02-29 21:53:45,509 - DEBUG - [Thread-4] USBPrinting.AutoDetectBaudJob.run [50]: Checking /dev/cu.usbserial-AH05LL5O if baud rate 250000 works. Retry nr: 0. Wait timeout: 3
```

The check for baud rate 115200 should succeed. The one for 250000 doesn't seem to ever complete.

**Additional information**
I can communicate with the printer manually with `picocom -b 115200 /dev/cu.usbserial-AH05LL5O`

rgov, 2020-03-01:
There appears to be a bug in PySerial where the second time it calls `serial.readline()`, the read timeout does not apply.

rgov, 2020-03-01:
Something looks really wrong with the return values from `Timeout.time_left()`. In the loop, we keep calling `select()` with time outs like:

```
selecting with timeout <serial.serialutil.Timeout object at 0x100810dd0> 2.999992543000001
selecting with timeout <serial.serialutil.Timeout object at 0x1008139d0> 2.9999926980000007
selecting with timeout <serial.serialutil.Timeout object at 0x100810dd0> 2.9999922859999995
selecting with timeout <serial.serialutil.Timeout object at 0x1008139d0> 2.9999931029999996
selecting with timeout <serial.serialutil.Timeout object at 0x100810dd0> 2.999992982
```

rgov, 2020-03-01:
I investigated the timeout issue with https://github.com/pyserial/pyserial/issues/487.  Cura is assuming that the `readline()` call will properly time out after 3 seconds. But this approach does not work when bytes are being read at the wrong baud rate—it only times out if a total of 3 seconds elapse between bytes.

rgov, 2020-03-01:
The second issue is that Cura expects `ok` and `T:...` to be in the same line, whereas they come across from this printer in two lines:

```
got line: b'ok 0\r\n'
got line: b'T:19.07 /0 B:18.33 /0 B@:0 @:0\r\n'
```

rgov, 2020-03-01:
(Why doesn't it detect the correct, lower baud rate? The first line read from my printer looks like this. Cura doesn't disregard data that was received before the test `M105` message.)

```
\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6\x80\xfe\x1e\x1e\x1e\xe6\x00\x18\x00\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6\x80\xfe\x1e\x1e\x1e\xe6\x00\x18\x00\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6\x80\xfe\x1e\x1e\x1e\xe6\x00\x18\x00\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6\x80\xfe\x1e\x1e\x1e\xe6\x00\x18\x00\xe6\x00\x98\x80\x183\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6\x80\xfe\x1e\x1e\x1e\xe6\x00\x18\x00\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xe6\x80\xfe\x1e\x1e\x1e\xe6\x00\x18\x00\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6\x80\xfe\x1e\x1e\x1e\xe6\x00\x18\x00\xe6\x00\x18\x00\x18s\xe6|\x1e\x7f\xe6|\xf8\x1e\xe0|\x18\x18\xe0\x18\x1ep\xc6\x80start\r\n
```

mahtDFR, 2020-03-02:
Thanks for the detailed report. Unfortunately there is no time in our road map to support this, so it is deferred for now. We are open to pull requests fixing this, though.

rrrlasse, 2020-09-29:
Try and see if https://github.com/Ultimaker/Cura/pull/8463 should fixes the baudrate issue