Ultimaker/Cura #7198: make the "pause at height" post process  context based

SchlSE, 2020-03-01:
**Is your feature request related to a problem? Please describe.**
The Post process list is in my opinion a mess. instead of having one entry for "pause of layer height"
we have four entries that aren't even listed directly after another. And the difference is basically the used commands. If you don't read every entry it is possible to use the false one, resulting in abortet prints ,ect.

**Describe the solution you'd like**
make it context sensitive to the printer and its code setting.

**Describe alternatives you've considered**
make a dropdown  menufor the version in the setting of the process

**Affected users and/or printers**
Basically everyone that uses Boards that doesn't use the marlin code set

fieldOfView, 2020-03-01:
I agree. Not only is it confusing for end-users which one to pick, but it is also a bit of a nightmare for maintenance, if eg an off-by-one error gets fixed in one flavor but not in the others.

fieldOfView, 2020-03-01:
I'm working on it here: #7201

Ghostkeeper, 2020-03-06:
I think most of the development team agrees wholeheartedly with this. It's been a pain to maintain just as well as it's been a pain to use. I'm glad fieldOfView took this on. We also have a ticket on our backlog about this.

PR #7201 (2020-05-19): Unify pause at height scripts
This PR adds an option to the pause at height script to set the pause method (gcode command), instead of having 4 different pause at height scripts. This should make maintenance of these script less work, since fixes don't have to be made in 4 scripts. It also gives feature parity to all the different flavors.

TODO:
* ~automatically suggest the right method for the current printer definition~ see a7e23ba
* ~add repetier-specific parking if repetier flavor is selected~ see c87ee61
* ~make "griffin" a method~ see 87633e7
* ~consider the need to upgrade configuratio [... cut]
closed 2020-05-19 by ninovanhooff