Ultimaker/Cura #7202: Enable retraction options when lift head option is enable

amioranza, 2020-03-01:
When using lift head enable I have some side effects related to material dripping from the nozzle and causing some strange artefacts (https://imgur.com/gallery/gouJEQa) on the printing. This occurs because when lifting no retraction is used, we don't have an option to enable it and can't define the retraction distance, that could be different from the main retraction settings.

My suggestion is to have two more options at the cooling section, one to enable retraction when lifting, and the later is to define the retraction distance when lifting the head.

Ghostkeeper, 2020-03-06:
It does retract during the lift though. Here's a snippet of g-code that shows a lift (made by loading a cube in and setting the minimum layer time to 100s). I added some comments to show what it's doing.

```
G1 F300 X120.273 Y103.939 E35.91491   ; Last line of previous layer.
M204 S5000
M205 X30 Y30
;MESH:NONMESH
G0 F600 X120.273 Y103.939 Z1.27       ; Move to next layer's Z to prevent scratching the model.
G0 F15000 X120.34 Y111.34             ; Move to next layer's starting position?
;Small layer, adding delay
G1 F1500 E29.41491                    ; <--- retraction of 6.5mm!
G0 F15000 X120.34 Y111.34 Z4.27       ; Move up 3mm.
G0 X106.197 Y97.197                   ; Move aside.
G4 P44827                             ; Pause.
;TIME_ELAPSED:513.794862
;LAYER:5
;MESH:cube.stl
G0 X120.34 Y111.34 Z1.27             ; First line of the next layer.
```

For this retraction it's using the normal retraction settings (speed and distance and unretract speed). Is there a good reason why this would be different from a normal retraction?

amioranza, 2020-03-08:
So, Do you have any idea of what is causing the extra printing things on my situation? It is strange because it appears to be the absence of retraction or something related to material dripping from the nozzle. At first I think this is a really good option to have this head lift in Cura, but in my case the results on some designs is are not so good as expected. About the retraction, I think different values allow more flexibility, for example, if I use 3 mm to normal operation and avoid string effect but to lift head I need more retraction then the value used to avoid stringing. In my case my printer has a bownden system, maybe I need more retraction, and to avoid stringing the 6mm value is ok. Maybe it could be related to the retraction speed, if I can set it just for head lift and not for regular retractions could good option.

All this suggestions is just a guess, I didn't had time to test more options of retraction and speed. So If you think it will not solve the problem you can close the issue.
Thanks.

Ghostkeeper, 2020-03-09:
Your problem is called "stringing" in 3D printing lingo. There are numerous articles online on how to prevent it. I've written an article too in the Settings Guide plug-in: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/stringing.md

iguanavon, 2020-03-10:
Hi Ghostkeeper,

I'm using Cura 4.5 but the gcode doesn't have the G1 command when performing lift head. 

I've played with the settings in the "Travel" tab, but none of it generate the G1 command for lift head. I've also deleted every thing in Cura folder and reinstalled Cura 4.5

May I know are there other settings that I need to enable in order to get the retraction for lift head?

[image]

[image]

Ghostkeeper, 2020-03-11:
The setting is called [Lift Head](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/cooling/cool_lift_head.md), under the Cooling category.

Lifting the head will only happen if:
* The layer is small enough to hit the Minimum Layer Time.
* The nozzle would need to move slower than the Minimum Speed in order to still reach the Minimum Layer Time.
* There is still some time leftover. It's going to lift the head and wait out this remaining time on the Minimum Layer Time.

Ghostkeeper, 2020-03-11:
I see in the source code that it's just making a normal retraction:

https://github.com/Ultimaker/CuraEngine/blob/dbf3d79a112894fdf9f3b9460923f85256f04b44/src/LayerPlan.cpp#L1727

The normal writeRetraction function creates a retraction if:

* Retractions are enabled.
* Retraction distance is more than 0mm.
* We haven't had too many retractions in the past. For your settings in your screenshot that is 90 retractions in the last 4mm of filament. Should be fine!

So from your settings I'd expect there to be a retraction unless it's had excessive amounts of retractions before the head lift.

Ghostkeeper, 2020-03-11:
Maybe a project file would help to debug what's going on.

iguanavon, 2020-03-12:
I was preparing the project file and gcode for debugging and I thought why not verify the actual print?

What I've observed in the actual print is that retraction was done prior to the hotend lifting up and moving away.
So in the gcode file, there was a G1 command just before the command to move the hotend.

I'm guessing that's why the G1 wasn't generated again for the lift head?

[image]

I've zipped the 3mf and gcode files in this attachment if you're interested in finding out why our gcode are generated differently.

[G1 generated before lift head.zip](https://github.com/Ultimaker/Cura/files/4322198/G1.generated.before.lift.head.zip)

amioranza, 2020-03-12:
Here is the snnipet of my GCODE generated by Cura 4.5:
;TYPE:SKIN
G1 F960 X124.499 Y134.116 E3714.28523
G1 X124.499 Y91.689 E3715.74127
G1 X124.545 Y91.452 E3715.74956
M204 S3000
M205 X10 Y10
G0 F6000 X124.705 Y91.329
M204 S1800
M205 X8 Y8
G1 F960 X124.669 Y91.293 E3715.7504
M204 S3000
M205 X10 Y10
G1 F3600 E3709.7504
;MESH:NONMESH
G0 F600 X124.669 Y91.293 Z27.3
G0 F6000 X126.165 Y91.293
G0 X126.153 Y135.877
G0 X122.847 Y135.877
G0 X122.835 Y134.496
G0 X124.1 Y134.496
;TIME_ELAPSED:5128.208473
;LAYER:135
M204 S1800
M205 X8 Y8
;TYPE:WALL-INNER
;MESH:anycubic_i3_mega_fan_duct_v3_lowprofile.STL
G1 F3600 E3715.7504
G1 F2400 X124.1 Y91.099 E3717.19379
G1 X124.9 Y91.099 E3717.2204
G1 X124.9 Y134.496 E3718.66379
G1 X124.1 Y134.496 E3718.6904
M204 S3000
M205 X10 Y10
G0 F6000 X123.7 Y134.896
M204 S1800
M205 X8 Y8

@Ghostkeeper can you help me to identify if it doing the right retractions?

This is my Travel setup in Cura:
[image]

Ghostkeeper, 2020-03-13:
> I'm guessing that's why the G1 wasn't generated again for the lift head?

Yup, that would be it. Cura won't make a retraction if it's already retracted. The G1 command would do nothing.

> can you help me to identify if it doing the right retractions?

There are no retractions in your g-code. From this snippet it's impossible to determine whether or not that's correct. It's unrelated to this issue though since there is no head lifting either.

closed 2020-03-13 by Ghostkeeper