Ultimaker/Cura #7203: "Skin overlap" does not apply to "Top surface skin layers"

woschtkopp, 2020-03-01:
If I use "Top surface skin layers" there are gaps between walls and the top surface skin layers.
I upped "Skin overlap percentage" to 50% only to discover that this does not apply to the "top surface skin layers"
Could you add a setting "Top surface Skin layers overlap percentage"
That or simply apply the skin overlap to the Top surface Skin layers

printingotb, 2020-03-05:
I have noticed this to.
In my case, I want to reduce it to prevent the overlap material to cause uneven surface near some types of edges.
I think it would be best if it was a separate setting, just like every other line type has it's own flow setting.
This would allow us to print with more overlap on top/bottom skin for added strength while also use less on top surface lines to get the smoothest surface possible. This also helps when using ironing, giving it a better surface to work on.

Ghostkeeper, 2020-03-06:
Duplicate of #5758.

closed 2020-03-06 by Ghostkeeper