Ultimaker/Cura #7204: Allow Flow Rate to be set Per Material

NameOfTheDragon, 2020-03-01:
**Is your feature request related to a problem? Please describe.**
Flow rate is best set per roll of filament. The recommendation is that for each new roll of filament, flow rate should be re-calibrated and the value written on the roll of filament.

It makes no sense therefore to set flow rate on a per-model basis. If I later reload a model and print it with different material or color, then the flow rate needs to change.

**Describe the solution you'd like**
I define different materials to describe each roll of filament material and colour. My assumption is that rolls of the same material and color should have the same flow rate.

Therefore, I would like a per-material flow rate setting. This will allow me to reload a model, change the material and automatically have the flow rate adjusted for that roll of material.

**Describe alternatives you've considered**
I looked into using the OctoPrint "filament manager" extension to do this, but unfortunately it doesn't offer a way to vary flow rate. It does offer per-spool temperature offsets, which is heading in the right direction, but nothing for flow rate.

**Affected users and/or printers**
Everyone (all users/all printers) stand to benefit.

**Additional context**
Video demonstrating flow rate calibration and recommending doing it for each new roll of filament. https://youtu.be/lH-RrjtiC8M?t=785

fieldOfView, 2020-03-01:
Check out the Material Settings plugin from the Marketplace. It will let you most settings to the Material pane of the preferences, saved per-material. Do note however that if you change the same setting value in the sidebar, that value will override what is set in the Material

Ghostkeeper, 2020-03-06:
Duplicate of #2058.

closed 2020-03-06 by Ghostkeeper