Ultimaker/Cura #7206: [4.5.0] Cura inserting M140/M104/M190/M109 gcode before printer start gcode

eduwes, 2020-03-02:
**Application version**
4.5.0

**Platform**
Windows

**Printer**
Ender 3

**Reproduction steps**
Adding the following to the start gcode used to heat bed and hot end simultaneously. As of 4.5.0 Cura now inserts it's own M140, M190, M104 and M109 beforehand making the below not function

M140 S80 ; begin heating bed
M104 S240 ; begin heating nozzle
M190 S80 ; Wait for bed temp
M109 S240 ; Wait for nozzle temp

fieldOfView, 2020-03-02:
Instead of hardcoding a number for the temperatures, you should be using the `{material_print_temperature_layer_0}` and `{material_bed_temperature_layer_0}` replacement patterns. If you do that, Cura will not add its own preheat sequence.

closed 2020-03-04 by eduwes

Sophist-UK, 2020-04-24:
Should you use `{material_print_temperature_layer_0,0}` rather than `{material_print_temperature_layer_0}` in order to get the settings temperature rather than the base machine temperature?

fieldOfView, 2020-04-24:
I could be wrong, but I think that if you don't specify an extruder nr for a setting that is settable per extruder, Cura picks the default extruder number.

Sophist-UK, 2020-04-24:
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_start_gcode.md

Ghostkeeper, 2020-04-28:
Sorry, that article is wrong here. I'll make a correction to the Settings Guide. After testing this, it appears it's indeed using the first available extruder.