Ultimaker/Cura #7207: Feature Request: UI to add additional nozzle sizes

MarcusWolschon, 2020-03-02:
**Is your feature request related to a problem? Please describe.**
yes, the "1.0mm" nozzle can not be selected/added.
I can work around that by changing "line width" every time but adding
nozzle sizes that Ultimater did not anticipate would be the clean solution.

**Describe the solution you'd like**
Add a button to "copy" additional nozzle sizes by copying an existing nozzle size, entering a diameter and saving.

**Describe alternatives you've considered**
A proper documentation of the complex file format, so people can add the missing entry themself.
(Probably has to be done in multiple places because materials may have different settings based on nozzle size.)

**Affected users and/or printers**
All printers where nozzles can be changed.

**Additional context**

fieldOfView, 2020-03-02:
This is going to be tricky, because the "nozzle selection" not only selects a nozzle size but it also selects a different set of (quality) profiles. For a new nozzle size, there will not be any profiles.

MarcusWolschon, 2020-03-03:
@fieldOfView That's why I suggest a "copy" button instead of a "new".
Create copies of the profiles of the user-selected, existing nozzle size.

**For clarification:**
Are these material profiles 
(change temperature tomake it soft enough for a small nozzle or cool enough to still bridge gaps despite thicker line-width)
or printer profiles
(reduce flow rate, so as not to overload the extruder motor with tiny nozzles)?

If it's material profiles, there will probably not be profiles for all suggested (or worse, hard coded)
nozzle sizes anyway and thus an extrapolation of missing profiles from existing ones.

fieldOfView, 2020-03-03:
> Are these material profiles or printer profiles

A bit of both. As an example, here's the set of quality profiles for the Ultimaker 2+, which has 4 nozzle sizes (0.25, 0.4, 0.6 and 0.8), 8 materials and 7 qualities, but not all permutations are supported (have profiles):
https://github.com/Ultimaker/Cura/tree/master/resources/quality/ultimaker2_plus

MarcusWolschon, 2020-03-04:
That's an insane rabbit hole you have maneuvered yourself into.
Given that materials come from plugins,
the number of printers explodes
and nozzles as well as entire hot-ends/print cores can come from third parties.

nallath, 2020-03-04:
> ...and here you try to have individual profiles for the matrix product of them all.

That's also why we don't try to do that. It's a sparse matrix.

Ghostkeeper, 2020-03-06:
I'd like to reject this honestly. Printer modifications are something we've always kept out of scope for the GUI. If you've made a modification to your printer, the step to writing a new .cfg file is not very big. Just copy the existing nozzles from the `variants` folder in the Cura installation and make modifications as you wish. Or else use a Custom FFF Printer and start from scratch.

We're aware that the number of profiles grows exponentially with the number of interchangeable modules in your printer, yes. We've limited this exponential growth in several ways (e.g. printers sharing their profiles, filtering quality profiles by material type rather than individual materials, etc) but it's still exponential. However changing this breaks a multitude of other requirements that we need to keep. Because yes indeed the printing temperature will need to change for 0.06mm layer height ONLY when printing PC through a 0.25mm nozzle or something. The configuration tree is sparse in that it doesn't require e.g. all material-nozzle combinations to have the same amount of quality profiles and have methods to filter the intersection profiles out for multi-extrusion printers.

Materials coming from third parties can set their material type to be the same as a similar material profile if they want to share the same quality profiles. In practice, most don't do this as there are always differences in the blends. Print cores can't share their profiles with other print cores at the moment.

MarcusWolschon, 2020-03-09:
A nozzle is not a modification of the printer as the printer is build to allow changing nozzles to other sizes.

Ghostkeeper, 2020-03-09:
Yes, Ultimaker's printers also have swappable nozzles. So they provide a drop-down with the nozzles you can choose from.

But if you'd want to add any nozzles you'd still need to manufacture your own nozzles which is one of the most complex things about building a printer. You need a pretty precise milling machine for that and a lot of theory about how polymers flow in different stages of their glass transition.

YiffyToys, 2020-03-09:
Or just buy a bag nozzles from a manufacturer.
e.g. the "Extra nozzles" from Ultimaker itself or an Olsson Ruby or any other one.
That doesn't make it a modification of the printer.
Just as exchanging the spool of material isn't a modification of the printer.
So the argument presented 3 days ago is invalid.

Ghostkeeper, 2020-03-11:
If you're buying a nozzle from someone else then someone else manufactured this nozzle but then didn't provide Cura profiles for it. They could share them with their buyers or provide them to Cura to ship with the releases.

Essentially this is what Ultimaker is doing too. You can buy nozzles by the bag (well, box I guess), so Ultimaker provides profiles for these nozzles.

MarcusWolschon, 2020-03-11:
@Ghostkeeper Interesting proposal.

1. The Marketplace allows for plugins that add Material profiles and to add printers but not plugins that add nozzles. That category doesn't exist.

1. How can I write, share and import profiles for nozzles? Are additional features required to do that?

1. What is included in that profile except for the line_width? Some kind of offset to apply to existing profile?

1. In your proposal, the 3rd party manufacturer of the nozzle (quite possibly the user himself with a drill or a lathe) wouldn't know about the 3rd party manufacturers of all materials any users may use.

**Summary**
I still think that the requested feature to copy an existing nozzle size or make an alias (and thus use  all existing material and printer combinations that refer that existing nozzle size) is the easiest solution for existing users. Finer, larger(faster) and in-betwen sizes will be similar enough to an existing size. The user just no longer has to override the line_width every single time (and to actually be educated by proper user documentation that this is the way to allow non-default sizes at the moment.).

MarcusWolschon, 2020-03-11:
@Ghostkeeper 
https://github.com/Ultimaker/Cura/wiki/Creating-Packages
There are no plugin types for nozzles.
Only " material (.xml.fdm_material) and quality (.inst.cfg) files. "

Ghostkeeper, 2020-03-12:
> The Marketplace allows for plugins that add Material profiles and to add printers but not plugins that add nozzles. That category doesn't exist.

Indeed the Marketplace has no place for custom nozzles yet. I don't think there is any qualms against this, but Ultimaker hasn't had the incentive for this either since there is only one single Chinese copy of their "print cores" out there and it's not necessarily something that Ultimaker wants to build a feature for. We were also planning to add a category for printers. It's still something we want to add at some point (with the incentive that it's some work and responsibility to maintain these profiles in the Cura repository) but it doesn't get any priority at the moment.

> How can I write, share and import profiles for nozzles? Are additional features required to do that? Where is the documentation for manufacturers?

To write nozzle profiles is the same as to write quality profiles, except for the metadata. You can take example from the existing profiles, e.g. https://github.com/Ultimaker/Cura/blob/master/resources/variants/ultimaker3_aa04.inst.cfg . These are installed by putting the files in your resources folder.

To share nozzle profiles I suggest a public version controlled repository or e-mail, depending on who you need to share it with and how often you'd need to update it. Other ways of sharing files are also fine by me, like pen and paper, smoke signals, shouting loudly, etc. They are just files. If you'd like to share them with the Cura user community at large, you need to share them via a pull request to our repository. We'll review them before we add them.

> What is included in that profile except for the line_width? Some kind of offset to apply to existing profile?

Line width is typically not included in the nozzle profile. What you'll probably want to overwrite is the nozzle size and the temperature/cooling parameters required to handle that. The line width gets automatically adjusted, by default to 7/8th of the nozzle size. Most likely you will also need to provide quality profiles (and perhaps material profiles) since different nozzle sizes will be printing at different layer heights. You can't realistically print 0.04mm layers using a 1mm nozzle.

> In your proposal, the 3rd party manufacturer of the nozzle (quite possibly the user himself with a drill or a lathe) wouldn't know about the 3rd party manufacturers of all materials any users may use. He/she would be unable to ever provide material-specific values because these are the result of tests by the material manufacturer.

If the material and nozzle are made by different people, you'll always have the question of who is going to create profiles for the combination. I don't really have a solution for this. At Ultimaker they create profiles for all nozzles for a subset of the materials that Ultimaker wants to support, and let other material manufacturers create profiles for their own materials using a subset of our nozzles. So it's really coming from both sides in this regard.

> How does the current resolution strategy for the sparse matrix work?

Cura doesn't care who made the profiles, whether it was a printer manufacturer, a nozzle manufacturer, a material manufacturer or the community at large. It also doesn't care how you installed the profile, be it with the Cura distribution itself, via the marketplace or by installing files into your resource folder manually. And I'm assuming here that "profiles for material X" also means "quality profiles specific to material X and some subset of the available nozzles".

In your example, Printer A will have a drop-down allowing you to select from the 0.1, 0.2, 0.3 and 0.5mm nozzles, and the materials list would contain all of the materials in Cura that have the correct filament diameter (e.g. 1.75mm). If you select PLA or ABS, you'll hopefully get the options of several quality profiles. If you select any other material, your material drop-down will become orange and you won't be able to select any quality profile (but the global defaults for those materials will still be applied as a starting point). So if you want to print nicely you'll want to start with the 0.5mm nozzle and a custom CF20 profile and tune your custom profile from there.

Similarly, Printer B will have a drop-down allowing you to select from the 0.1, 0.3175 and 0.4mm nozzles. If the filament diameter is the same as Printer A you'd see the same list of filaments. But this time, the quality profiles may be different so you'd see the material drop-down become orange for different filaments, depending on which materials you installed. Cura won't allow the same material ID to be used multiple times, so there can't be multiple PLA material profiles. They can have the same user-visible names though, so if there is a name clash you might see "PLA" in the list twice. The quality profiles are shared between materials with the same material type, so if both PLAs have the same material type you'd be able to select between the quality profiles of both regardless of which material is active at the moment.

I'm not sure what answer you are hoping for but this is the best I'm able to give in reasonable time.

MarcusWolschon, 2020-03-12:
I wasn't talking about "print cores". These models are unaffordable.
I was talking about nozzles. Olsson Block nozzles.

MarcusWolschon, 2020-03-12:
"You can't realistically print 0.04mm layers using a 1mm nozzle."
I can print 0.04mm (1 full step) layers with a 1mm nozzle way better (and do so regularly) then with an 0.4mm nozzle because my extruder stepper for 2.850mm material is still turning very slowly and not practically stopped.

MarcusWolschon, 2020-03-12:
"I'm not sure what answer you are hoping for"
Sadly it's not.
The question was how the SPARSE matrix is handled.
How it handles combinations that have no specific profile.
If it looks for the same material in a different nozzle size or the same nozzle size in a different matertial (but the same "material type" as you put it) and if it uses that value or averages multiple neibouring values based on some kind of distance metric in the matrix.
There has to be an order. A priority. To handle resolution of missing combinations.
Else such combinations would be just inpossible to slice.

Ghostkeeper, 2020-03-17:
> The question was how the SPARSE matrix is handled.
How it handles combinations that have no specific profile.

If there is a configuration of printer, material and nozzle that has no quality profile, it'll make the material selection yellow, indicating that the configuration is not supported. It will load in no quality profile, showing only the default setting values for that combination of printer, material and nozzle. It does not look in the profile tree for different nozzles or different materials, and doesn't do any averaging or anything. There is no priority or order. Just an indication that there is no profile.

These combinations are not impossible to slice. It'll just revert to the defaults as closely as it can without going into specifics of the neighbouring configurations.

MarcusWolschon, 2020-03-17:
Thanks.
So the proposed profiles for new nozzle sizes provided by a 3rd party making/modifying/selling the nozzle would never work for any 3rd party material.

So my suggestion to offer a UI to copy from a user-selected, existing nozzle size stands.

Ghostkeeper, 2020-03-17:
Side-conversations notwithstanding, I'm sticking with the same argument we've always held there and which I mentioned before. Making modifications to hardware is not something we'll support in the GUI. It doesn't really matter if you made those modifications, or Mr. Anders Olsson did and you're just buying them off of him.

If someone makes alternative nozzles (be it you or your neighbour or Olsson or Creality or whatever), they can only come in the list of nozzles by creating nozzle profiles with a text editor and adding them to the Cura resources.

closed 2020-03-17 by Ghostkeeper

MarcusWolschon, 2020-03-17:
Documentation at 
https://github.com/Ultimaker/Cura/wiki/Definition-Files-Explained
doesn't document anything regarding variants to actually do this.
Not even where to store them, so they survive an upgrade of Cura or the machine definition.
(Variants are aparently not stored in "user", but that seems to be the only folder to modify anything in.)
Help menu also links no documentation of the profiles and no developer documentation.

Ghostkeeper, 2020-03-19:
Yes, our documentation on this is lacking. There are plenty of examples though. Variants are in the "variants" folder: https://github.com/Ultimaker/Cura/tree/master/resources/variants (as opposed to user changes which are in the "user" folder). They are mostly the same as the other types of profiles, only the "type" metadata is set to "variant" so that it gets recognised, and they are not specific to material or variant so they just specify a definition that they are specific to. The `hardware_type` that you see in some profiles should default to `nozzle` if it's not specified.