Ultimaker/Cura #7209: Support around round part is creating a solid perimeter that keeps from being removed.

ForceConstant, 2020-03-02:
**Application version**
4.5.0

**Platform**
Win64

**Printer**
Ender3

**Reproduction steps**
I am trying to print a special gear, that has a top that needs support. The problem I have is the support that is being generated isn't able to be removed as it creates a continuous circle around part. Any ideas on how to print this successfully? I tried all the different support types, but they all seem to make the same outer shell around the support.

**Screenshot(s)**

[image]
[image]

**Actual results**
Zig Zag pattern is encased in a circle that makes it not able to remove support.

**Expected results**
Have a zig zag pattern that is easy to remove.

**Project file**
TBD

**Log file**
TBD

mahtDFR, 2020-03-02:
Not really sure if this can be called a bug but I've put it up for discussion.
What if you change the support settings to make support easier to remove? Can you share a project file please? (File > Save as...) so I have your settings.

ForceConstant, 2020-03-02:
> Not really sure if this can be called a bug but I've put it up for discussion.
> What if you change the support settings to make support easier to remove? Can you share a project file please? (File > Save as...) so I have your settings.

[CE3_Hex Cap Gear.zip](https://github.com/Ultimaker/Cura/files/4278352/CE3_Hex.Cap.Gear.zip)

mahtDFR, 2020-03-03:
As a specific workaround for your situation, I'd suggest setting support walls to '0'. This means support walls won't be printed, only support infill, so you can grab the infill with pliers easily to remove it. Here's the result I got with your project file in 4.5. I've also attached a 3mf for you here.
[CE3_Hex Cap GearMJ.zip](https://github.com/Ultimaker/Cura/files/4280948/CE3_Hex.Cap.GearMJ.zip)

[image]

I don't think we're going to take this further as a feature, so I'll close this ticket. But thanks for your suggestion!

closed 2020-03-03 by mahtDFR

ForceConstant, 2020-03-03:
> As a specific workaround for your situation, I'd suggest setting support walls to '0'. This means support walls won't be printed, only support infill, so you can grab the infill with pliers easily [...]

Thanks. Yes I didn't know about support walls,