Ultimaker/Cura #7210: Color differentiation in settings panel

vihrovart, 2020-03-02:
**Is your feature request related to a problem? Please describe.**
When I trying to find some setting (among hundreds), it's very difficult without color differentiation. In other slicers i can find groups of settings on different tabs. But in Cura i don't. It's sad and difficult.

**Describe the solution you'd like**
Color differentiation in settings panel (in advenced mode). For example, settings group would have some color marker, may be in left side of name of setting.
I think it would be very useful, for new users like me, in my investigations of settings.

**Affected users and/or printers**
All users

mahtDFR, 2020-03-02:
Do you imagine this to be per category? e.g. the Quality settings category is a different colour from the Shell settings category? Should each category have a distinct colour? Or alternating colours? It may have some conflicts with our existing style choices.

vihrovart, 2020-03-02:
I mean color marker on the left side of each setting, linked for each category (may be custom, user can set color for marker in properties for each setting or each category, for example, in best case).

Situation:
I set some setting in experemental category, and then i want to change flow setting, or other one, and i begin scroll and scroll up settings panel, and it's very difficult to find the setting what i wont. I have to read all of them, because i don't remember where is flow (e.g.) setting. But, if i wold have some marker (e.g. green) on the left side of all settings in materal category (and other in other sections), i wold remeber, that flow setting in "green category" (in my memory i wold have color association with flow setting), and i would scroll to "green section", and than easy and quick would find what I want.

vihrovart, 2020-03-02:
This marker can be like colored square, or colored circle near setting's name

fieldOfView, 2020-03-02:
>  and then i want to change flow setting, or other one, and i begin scroll and scroll up settings panel,

Try the search field.

vihrovart, 2020-03-02:
Settings it's main instrument in Cura (I think), and it would be great, if users could quickly navigate :-)

> > and then i want to change flow setting, or other one, and i begin scroll and scroll up settings panel,
> 
> Try the search field.

The search work very slow... And i use this feature in some cases, but color marker is useful and easy

I made gif for illustrate slow search
[https://ibb.co/NpyRwtk](url)

fieldOfView, 2020-03-02:
Adding more decoration to the settings will not make Cura any quicker.

vihrovart, 2020-03-02:
Color marker it is not decoration, but visual orientation tool

fieldOfView, 2020-03-02:
A good way to make the search much faster is *not* to make all settings visible. It will also make it easier to find back settings. Make a set of the settings you use for 90% of your prints, and then use the search to find the others when you need them.

mahtDFR, 2020-03-03:
@vihrovart thanks for the suggestion. We discussed this in our stand up and don't feel this feature is necessary for us to devote development time to right now. There are already implemented colourblind assist themes which can be accessed from preferences.

closed 2020-03-03 by mahtDFR

Ghostkeeper, 2020-03-06:
This feature request is not about colour blindness though. It's about identifying a setting more easily by giving each setting a unique colour.