Ultimaker/Cura #7213: [4.5.0] Cura crashes immediately, "Python error in main script"

Aaron-Hartwig, 2020-03-02:
**Application version**
Cura 4.5.0

**Platform**
```
------------------
System Information
------------------
      Time of this report: 3/2/2020, 10:33:17
             Machine name: G4HTZ1X2E
               Machine Id: {C3EB1AC4-5A51-412F-A3DF-6889229A6B77}
         Operating System: Windows 10 Pro 64-bit (10.0, Build 17763) (17763.rs5_release.180914-1434)
                 Language: English (Regional Setting: English)
      System Manufacturer: Dell Inc.
             System Model: Precision 7730
                     BIOS: 1.12.1 (type: UEFI)
                Processor: Intel(R) Xeon(R) E-2176M  CPU @ 2.70GHz (12 CPUs), ~2.7GHz
                   Memory: 65536MB RAM
      Available OS Memory: 65296MB RAM
                Page File: 18144MB used, 56879MB available
              Windows Dir: C:\windows
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
         User DPI Setting: 96 DPI (100 percent)
       System DPI Setting: 96 DPI (100 percent)
          DWM DPI Scaling: Disabled
                 Miracast: Available, with HDCP
Microsoft Graphics Hybrid: Supported
           DxDiag Version: 10.00.17763.0001 64bit Unicode

------------
DxDiag Notes
------------
      Display Tab 1: No problems found.
      Display Tab 2: No problems found.
        Sound Tab 1: No problems found.
          Input Tab: No problems found.

--------------------
DirectX Debug Levels
--------------------
Direct3D:    0/4 (retail)
DirectDraw:  0/4 (retail)
DirectInput: 0/5 (retail)
DirectMusic: 0/5 (retail)
DirectPlay:  0/9 (retail)
DirectSound: 0/5 (retail)
DirectShow:  0/6 (retail)

---------------
Display Devices
---------------
           Card name: Intel(R) UHD Graphics P630
        Manufacturer: Intel Corporation
           Chip type: Intel(R) UHD Graphics Family
            DAC type: Internal
         Device Type: Full Device (POST)
          Device Key: Enum\PCI\VEN_8086&DEV_3E94&SUBSYS_08321028&REV_00
       Device Status: 0180200A [DN_DRIVER_LOADED|DN_STARTED|DN_DISABLEABLE|DN_NT_ENUMERATOR|DN_NT_DRIVER] 
 Device Problem Code: No Problem
 Driver Problem Code: Unknown
      Display Memory: 32775 MB
    Dedicated Memory: 128 MB
       Shared Memory: 32647 MB
        Current Mode: 1920 x 1080 (32 bit) (60Hz)
         HDR Support: Not Supported
    Display Topology: Internal
 Display Color Space: DXGI_COLOR_SPACE_RGB_FULL_G22_NONE_P709
     Color Primaries: Red(0.646484,0.337891), Green(0.316406,0.614258), Blue(0.157227,0.043945), White Point(0.313477,0.329102)
   Display Luminance: Min Luminance = 0.500000, Max Luminance = 270.000000, MaxFullFrameLuminance = 270.000000
[... 144 lines cut ...]
         MPO Formats: Not Supported
    PanelFitter Caps: Not Supported
 PanelFitter Stretch: Not Supported
```

**Printer**
Never was even able to open application.

**Reproduction steps**
1. Tried to open Cura.

**Screenshot(s)**
[image]

**Actual results**
Cura does not open.

**Expected results**
Cura would open.

**Project file**
Not applicable.

**Log file**
No log file generated. Seems like the application crashes before it can make one.

Ghostkeeper, 2020-03-06:
Oof, this is a bit worrying. It's crashing when it initialises the crash reporting module...

Ghostkeeper, 2020-03-06:
Something wrong with a network proxy, maybe?

7uxx, 2020-03-12:
Hey,

got the same error:

[image]

Tried to use with disabled network and got the same error again. Deleted every config and cache folders in %APPDATA% and %LOCALAPPDATA% and reinstalled with both .exe and .msi installers, didn't helped.

Ghostkeeper, 2020-03-17:
That is not the same error as Aaron-Hartwig showed though.

Aaron-Hartwig, 2020-03-17:
> Something wrong with a network proxy, maybe?

My proxy configuration works for any other tools I have that require network access. When I attempt to do this from my home network (no proxy) I still get the same error.

7uxx, 2020-03-17:
> That is not the same error as Aaron-Hartwig showed though.

This IS exactly the same error. 

Just loaded with CLI, thats why it looks different.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Aaron-Hartwig, 2024-11-22:
No clue. Feel free to close it.

closed 2024-11-22 by GregValiant