Ultimaker/Cura #7215: [4.5.0] Weird duplicate errors in print

FlailAway, 2020-03-02:
**Application version**
4.5.0, same for 4.4.1

**Platform**
Mint 19.3+Cinnamon

**Printer**
Anycubic Delta
MKS Gen-L
Marlin 1.1.9

**Reproduction steps**
Print it inside radii.

**Screenshot(s)**
[image]
[image]

What is happening here? I first printed just one of these things and it had the gaps on the inner radius. 

I rotated it on the build plate to see if it was the Delta, sliced and printed again and same problem.

I then tried putting two on the build plate and rotating one through 90-degrees, (second pic) but still get seemingly identical results on the inside radii.

I then re-drew the inside radii to flats and printed and no problems at all, perfect smooth and flat surface. I used the same Profile and settings for all prints. 

The Preview shows a perfect curve with no gaps but that pattern is being repeated no matter what while it is printed as an inside radii.

smartavionics, 2020-03-02:
You know what I'm going to ask. Please provide the project file, without that, it's impossible to really know what's happening. Thanks.

FlailAway, 2020-03-02:
How do I make a project file? Do you want the .obj and an exported profile or what? Just printing one using PrusaSlicer and it is coming out perfect, but not a fan of PrusaSlicer so would like to get this working with Cura.

smartavionics, 2020-03-02:
It's easy. Just load the models onto the build plate and have the correct profile selected and then do File -> Save from the menu bar. Zip the resulting .3mf file and attach the zip to this issue.

FlailAway, 2020-03-02:
OK, are you then able to print from that file? I cannot give away the project items, they are not mine to do so and doubt I'd get permission to either. Especially so if I am adding them to a post here for anyone to DL.

smartavionics, 2020-03-02:
OK, leave the models out and just save an empty project that has the same profile and attach that. At least I can then look at the settings even if I don't have the model. Thanks.

FlailAway, 2020-03-02:
I made a small testpart where the errors are. I have not printed the test piece as the Prusa sliced ones will do fine for now.

Don't have .zip, so renamed it .log.

[test-01.log](https://github.com/Ultimaker/Cura/files/4278153/test-01.log)

smartavionics, 2020-03-02:
Thanks for the file. It helps a lot.

So, the reason you get those ugly gaps in the outer wall is because you are getting a long(ish) un-retracted move at the end of each layer and so when it starts the outer wall at the beginning of the following layer, the filament has oozed from the nozzle leading to underextrusion.

You can do various things to avoid this.

1 - enable retract at layer change.

2 - don't print the outer wall first.

3 - set the max comb distance with no retract to a low (but non-zero) value. e.g. 5

Another thing is that the z-seam is hopping around because you are using the sharpest corner for z-seam alignment. It's only really any good for a small set of model shapes and I would suggest you use the "user specifed" setting as that will keep the z-seam from jumping about.

Hope this helps.

smartavionics, 2020-03-02:
I also recommend setting connect infill lines as this will help keep the nozzle from emptying.

FlailAway, 2020-03-03:
OK, thanks -- but of those 4-things only 1 (#2 above) is set as you suggest in the stock Draft profile. I used the stock profile as my base before setting 120mm/s speeds for the Delta,  assuming you guys would know way more about the settings than me.

I will try a print with those settings, thank you, but if they are optimal, why are they not set in the stock Draft Profile? Regard that as rhetorical. :)

smartavionics, 2020-03-03:
I do not know why the default settings are so unhelpful. Someone from UM will have to comment on that.

FlailAway, 2020-03-03:
Thing is printing now, but after the changes you suggest, the outside diameters are now lumpy. It is a consistent pattern about 2mm pitch on the surface. But, the weird stuff has gone. Seems I have exchanged one issue for another. {grin}

I will have to revert to PrusaSlicer for now, but will look in on Cura in a few more iterations, it seems to have lost it's way a little -- again, shades of 3.6??

I fully understand the complexities of programming this stuff, but I need things to print with minimal interference from me.

One other thing, the Help fly-out for "Infill Pattern" has contradictory information for CUBIC, QUARTER CUBIC etc per layer/each-layer. Not important, just thought I'd mention it for the people handling the help stuff. (my CAPS to emphasize "cubic," etc)

FlailAway, 2020-03-03:
Sorry, my mistake. The 2mm-pitch lumps were FreeCAD resetting my circle segments, it is doing it with Prusa too.

Ghostkeeper, 2020-03-06:
> I will try a print with those settings, thank you, but if they are optimal, why are they not set in the stock Draft Profile? Regard that as rhetorical. :)

Our settings all have advantages and disadvantages (well, apart from Skirt/Brim Acceleration maybe). For many of the oldest settings the global defaults were tuned by the programmers of that time (i.e. Daid) and later optimised only for Ultimaker printers since that is the only printer we've tested them on. For newer settings the global default is sometimes set to make minimal changes from the previous g-code. Or a new feature is introduced that does change the defaults but then other defaults are less than optimal any more but we don't see that because it's only an issue for Prusa i3 printers or something.

If you have better print quality with certain settings, and another Prusa i3 user can confirm this, we'd be glad to merge those changes in. Perhaps @stelgenhof is willing to help?

stelgenhof, 2020-03-08:
@Ghostkeeper @FlailAway Sure, I can help as I can print on a Prusa MK2.5, Ender-3 for example.

Ghostkeeper, 2020-03-09:
Can you try printing something that has inner curves and retractions (such as [this model](https://github.com/Ghostkeeper/SlicerTestModels/blob/master/my/washing_machine_drain.scad)) and see whether it is improved by enabling "Retract at Layer Change" and/or setting the "Max Comb Distance With No Retract" to something small-ish like 5mm? I think the second one is the most influential one.

stelgenhof, 2020-03-11:
@Ghostkeeper @FlailAway In the process of printing the samples. Will let you know about the results as soon as possible.

stelgenhof, 2020-03-12:
Hi @Ghostkeeper @FlailAway 

Here are my results of printer a model with inner curves and retractions:
A. Base (Before changes)
B. With "Retract at Layer Change"
C. With "Retract at Layer Change" and "Max Comb Distance With No Retract" set at 5mm.

[image]
[image]
[image]
[image]

Sliced with Cura 4.5 using the default Draft (0.2mm) settings and printed on an Ender-3. The filament is not of the highest quality and perhaps a bit moist. 
Firmware: Klipper

At the bottom inner radius there are some small holes visible at the start of a layer. However these are noticeable on all 3 models. Visually all 3 models look pretty much the same and I didn't notice any difference between them.

Hopefully this helps.

Cheers! Sacha

Ghostkeeper, 2020-03-13:
Thanks. So basically the conclusion is that the bug can't be reproduced in the first place. Then merging that change to the defaults is maybe not such a good idea.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

stelgenhof, 2024-11-23:
I haven't used the latest version of Cura so can't say if this is still an issue. As far I am concerned, this issue can be closed.

closed 2024-11-23 by GregValiant