Ultimaker/Cura #7216: [4.5.0] Post Processing Plugin: "Pause at Height" retract problem

ilyko96, 2020-03-02:
The problem with 'Pause at Height' script for 'Post Processing Plugin' involves extra-extrusion after printing had been resumed. As far as I can understand, script only tracks last state of extrusion even if last 'extrusion line' was the retract. In this case script assumes that at the end (of script) extrusion has to be (relative) 0, while it has to be (relative) -retraction_value. More on reproduction of this situation below.

**Application version**
4.5.0

**Platform**
Windows 10 x64

**Printer**
Creality Ender 3

**Reproduction steps**
1. Load STL file from archive [board_main.zip](https://github.com/Ultimaker/Cura/files/4277186/board_main.zip)
2. Change _Print Settings_ as shown on Screenshot (key settings are highlighted in yellow)
3. Add 'Pause at Height' script in Post Processing plugin with settings as shown on Screenshot.
4. Slice and 'Save to file'
5. Check 'Actual results' and 'Expected results' sections

**Screenshot(s)**
[image]
[image]

**Actual results**
[image]
_Actual results description:_
* (1) and (2) are the last moves with extrusion.
* (3) - retract: 534.27547 - 4.0 = 530.27547
* Yellow lines - extruder related lines produced by script: retract 7mm -> extrude 12mm
* (4) - script sets extrusion to value that was right before gcode injection
* (5) - ordinary gcode undoes retract value (which leads to 4mm of extra extrusion noticeable as a large blob of plastic on the model)
* (6) - first move of next layer with extrusion

**Project file**
Project file can be found in [board_main.zip](https://github.com/Ultimaker/Cura/files/4277186/board_main.zip)

Ghostkeeper, 2020-03-09:
Yeah, I can see from the g-code that this is going wrong. Very clear explanation, thanks! I'll put this up with the other developers to discuss.

mahtDFR, 2020-03-09:
I've added this to our backlog: CURA-7288

iRoc999, 2020-04-13:
I also came across this issue, and was wondering if in the fix we could get access the the second retraction distance as a settable field. Feels like bowden drive systems can handle that no issue, but with direct drive if you set the retraction to fully retract. the second retraction causes jams ( at least on my printer every time) 
the ability to set the second retraction lower would be a life saver. Currently doing this by editing the GCode

Thanks