Ultimaker/Cura #7217: new to dual colour scripts ... need better solution to purge tower/bucket

leshuguets, 2020-03-02:
Purge towers incredibly wasteful and slow ... buckets are messy and don't work if print head can't go beyond bed dimensions. So an economic and fast solution is to print a simple purge line only on a tool change.
Its simple to alter the Tool change scripts in Cura to get it to print a simple purge line every time a tool change happens but we need to know the total number of tool changes in the Gcode  so that the Z height can be incremented for each new waste line. All that is needed is access to a variable e.g. V[tool_change_number_v] that holds count of total number of tool changes ... this should be simple to incorporate when the Gcode is compiled.

mahtDFR, 2020-03-06:
If I understand correctly, the bed would need to return to z0 at each tool change to make a purge line. Have I got that right? Where would the lines be placed? Can you clarify this a little?

Ghostkeeper, 2020-03-09:
We can't fill in this variable at the moment since the key replacement is done before slicing. It needs to be done before slicing, since we need to know whether or not to include temperature commands at the beginning of the print. Before slicing, it is unknown how many extruder changes there will be, as this depends on the actual shape of the models for all extruders (and the slopes of the surfaces for adaptive layers).

Your solution is possible in theory, by doing some key replacement at the end of the print as well, but impractical in practice. It will also pose issues with the print head or gantry colliding with previously printed objects. For this reason I'll reject this feature request.

closed 2020-03-09 by Ghostkeeper

leshuguets, 2020-03-09:
I understand the problems you pose but all could be solved by a simple 2
pass slice.
the first to count the number of extruder changes and positioning  for the
waste tower  the second to implement.
since the process is fast there would be no noticeable lag in slicing and
the benefits would be enormous

CheevRS

*_____________________*
*......Mike & Dorrie Norris......*
*.............Les Huguets...............*

*........46310 Frayssinet...........*
      www.leshuguets.co.uk

Ghostkeeper, 2020-03-13:
Yeah, a post-processing script could generate this in theory, as long as a safe position can be found to get the nozzle closer to the build plate.