Ultimaker/Cura #7218: No support of certain parts of print when slicing

Cronic-se, 2020-03-03:
**Application version**
(The version of the application this issue occurs with.)
**4.5.0**

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
**Windows 10, Nvidia GeForce GTX 1070Ti**

**Printer**
(Which printer was selected in Cura?)
**Ultimaker 3**

**Reproduction steps**
**1. Add my .stl-file in Cura
2. Print Core 1: Generic PVA, BB 0.4 | Print Core 2: Generic PLA, AA 0.4
3. Profile: Normal 0.15mm
4. Slice**

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
**Layer 3:**
[image]

**Layer 4:**
[image]

**Actual results**
(What happens after the above steps have been followed.)
As you see, Cura decides to skip support of certain parts of my print. I have tested different settings with no better result. The pictures above was sliced with profile: "Normal 0.15".

**Expected results**
(What should happen after the above steps have been followed.)
I Expect that Cura should ad support for all parts of my print.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
**http://zeal.se/UM3_Elektronik_2.1.3mf**

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log](https://github.com/Ultimaker/Cura/files/4279558/cura.log)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-03-03:
Often, that kind of problem is because the bottom of the model is not flat but I looked at the model and the areas that don't have support did look flat (as far as I could tell). However, as a workaround, I increased the initial layer height to 0.3 and it now fills in the missing support. Obviously, changing that layer height could cause other problems (YMMV).

[image]

Cronic-se, 2020-03-04:
> Often, that kind of problem is because the bottom of the model is not flat but I looked at the model and the areas that don't have support did look flat (as far as I could tell). However, as a [...]

Thanks for your reply!
I also found another workaround. I lifted the model up 0.4 mm to add another layer PVA. That fixed the problem.

Ghostkeeper, 2020-03-09:
For now, set the Support Stair Step Height to 0. That fixes the problem for you. We have some bugs with that stair stepping algorithm.

This is a duplicate of https://github.com/Ultimaker/Cura/issues/6379.

closed 2020-03-09 by Ghostkeeper