Ultimaker/Cura #7219: Make default brim extruder no. 2 if all objects are sliced with extruder 2 only

Dim3nsioneer, 2020-03-03:
If all objects on a dual extruder build plate are set to be printed with extruder two, the brim is still printed with extruder one. This leads to unnecessary preparation time.

I suggest the default brim extruder is set automatically to extruder two if only extruder two is used for any object prepared in the current slice.

An alternative could be a context menu option "set model+helpers to extruder X".
Another alternative could be to offer configurations in the quick material selection menu (used for network-connected Ultimaker printers) that only include one material but set the other one to "no material".

Single extrusion slices on dual extrusion printers will benefit from a significantly shorter preparation time, especially the Ultimaker S5 pro bundle.

This feature request is based on the situation in Ultimaker Cura 4.5.

mahtDFR, 2020-03-05:
This is on our backlog as CURA-7244

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.