Ultimaker/Cura #7222: Materials List gets corrupted

soleshoe, 2020-03-04:
**Application version**
4.5.0

**Platform**
macOS 10.15.3 (19D76)
macbook pro (early 2013), Quad-Core Intel Core i7, 16GB RAM
NVIDIA GeForce GT 650M
Intel HD Graphics 4000
Mouse: Logitech MX Anywhere 2

**Printer**
Tronxy X5SA PRO (2020)

**Reproduction steps**
1. go to "manage materials"
2. scroll through the list several times using the Logitech Free wheel mode

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
The materials list quickly get corrupted though all generic materials seems to be installed in the marketplace materials tab
Scrolling in list is erratic
I noticed the exact same problem in other scrolling lists
I noticed that the problem doesn't occur if I only use my trackpad to scroll
Disabling the freewheel mode seems to avoid the scrolling problem
Restarting cura doesn't fix the problem
Reinstalling cura doesn't fix the problem

**Expected results**
Should scroll smoothly through the list, not altering its content.
Since this problem occured, I only see 1 material available : silver metallic ABS (though all generic materials still seem to be installed).
**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4286685/cura.log)

**Additional information**
(Extra information relevant to the issue.)
[image]
[image]
[image]

nallath, 2020-03-04:
Could you share your configuration folder?

soleshoe, 2020-03-04:
> Could you share your configuration folder?

Which folder are you talking about ? Entire "Application Support / Cura" folder ?

Yian-Lai, 2020-03-04:
You can find the configuration folder by opening Cura>Help(in the menu)> Show configuration folder

soleshoe, 2020-03-04:
> You can find the configuration folder by opening Cura>Help(in the menu)> Show configuration folder

[4.5.zip](https://github.com/Ultimaker/Cura/files/4288322/4.5.zip)

soleshoe, 2020-03-04:
Note: I have erased the configuration folder and reinstalled Cura, avoid using free wheel mode with my mouse, and Cura is working fine.

Ghostkeeper, 2020-03-09:
Sorry, I've deleted the reply containing your configuration folder since it also contains the cached log-in information for your account, allowing anyone to log in. There are few things you can really do from your account within Cura (as Cura doesn't have the rights to change the password or anything) but still for security I've censored it. I'm keeping a copy of this folder for a few days in case Nallath (also an Ultimaker developer) wants to take a look at it, but I've logged out of your account to keep it secure.

Ghostkeeper, 2020-03-09:
With your configuration folder the problem reproduces. However I don't think that this is the result of excessive scrolling. The list of materials is empty because for some reason your extruder says that it's compatible with filaments of 175mm diameter (17.5cm). This is excessive. I think you meant it to say 1.75mm.
[image]
If you change that, the list of materials is restored.

closed 2020-03-09 by Ghostkeeper