Ultimaker/Cura #7224: Enlarge Machine Settings Dialog

thesec, 2020-03-04:
With version 4.5 the start and end gcode's text fields became veeeeery small - see attached screenshot.
Please enlarge / allow scrolling / allow to enlarge the dialog itself ...

[image]

tech2077, 2020-03-04:
This is also the case in 4.5 on MacOS. The window is too small and cannot be resized.

[image]

mahtDFR, 2020-03-05:
Thanks for the report.
@thesec which OS are you running here? MacOS as well?

smartavionics, 2020-03-05:
I agree the initial size of those gcode boxes is way too small. However, on Linux, it is possible to resize the whole dialog like any other window.

thesec, 2020-03-05:
@mahtDFR - yes, am running OS X, too.

fieldOfView, 2020-03-05:
I propose that instead of enlarging the window (which is a sliding slope), the gcode snippets are moved to a different tab, or they are edited using a widget that pops up a dialog.

thesec, 2020-03-05:
@fieldOfView - please keep the extruder's gcodes on mind when you decide to take it that way to keep it consistent.

nallath, 2020-03-06:
The issue was that the dialog still had hardcoded values in them (instead of using the dynamic sizing used by the rest of the code).
It also had it's min & max size set to be the same, which linux and windows interpreted as "Resize it to whatever you want" and osx as "No resize allowed". 
I've fixed both, so it shouldn't be needed to move the gcode to a different tab (anyway, I hope it won't be needed, because that's a lot more work)

closed 2020-03-06 by nallath