Ultimaker/Cura #7225: Wipe Nozzle Between N(!) Layers

thesec, 2020-03-04:
Please add a setting allowing a user to set the count of layers at which the wipe script is to be executed.

thesec, 2020-03-04:
While at it - where to find and edit the script?

mahtDFR, 2020-03-05:
> where to find and edit the script?

You'll find post processing scripts here (and at the same location in your install)
https://github.com/Ultimaker/Cura/tree/master/plugins/PostProcessingPlugin/scripts

I don't think this request fits in with our roadmap, but contributions are welcome. A post processing script would be a good way to implement it IMHO.

thesec, 2020-03-05:
Thank you, but that does not sound like any match for the "Wipe Nozzle Between Layers" setting what you seem to be referring to. Seems like "script" brought you up that twig, so I edited the description accordingly.

thesec, 2020-03-05:
... and am looking for the location of what's being referred to as "wipe G-Code" in that setting.

mahtDFR, 2020-03-05:
Yep, no script exists yet, but if someone created a script for this it would be a good way to implement it. You could consider adapting one of the existing scripts for this.

thesec, 2020-03-06:
You still seem on the wrong track. None is referring to any script any longer @mahtDFR !

@nallath, did you defer it according to @mahtDFR misinterpretation or other?

mahtDFR, 2020-03-06:
It's deferred because this is not a priority in our roadmap right now. I suggested that it can be achieved with a post processing script, but we don't have the resources to make one now.

thesec, 2020-03-06:
It will be a lot easier to modify the UI and have it writing the `;WIPE_SCRIPT_BEGIN ... ;WIPE_SCRIPT_END` only every nth layer which is what the request is about.

thesec, 2020-03-06:
Would still love to know where the few lines of wiping gcode come from ... does not seem an easy find :/

thesec, 2020-03-06:
Found the "wipe script" hardcoded in the engine's gcode export now.

Ghostkeeper, 2020-03-09:
The "wipe script" was originally a post-processing script provided by DesktopMetal. It's been integrated in CuraEngine since. See also https://github.com/Ultimaker/CuraEngine/pull/1000 . This is why people are confused, I think.

thesec, 2020-03-09:
That's explaining, yes.

closed 2023-07-13 by github-actions[bot]