Ultimaker/Cura #7231: [4.4.1, 4.5.0] Failure joint on small connecting pipe

aadhoc, 2020-03-06:
**Application version**
4.4.1, 4.5.0 (probably earlier too)

**Platform**
Windows 10 x64

**Printer**
Anycubic i3 Mega

**Reproduction steps**
1. Load STL file ([from this link](https://drive.google.com/file/d/14gOY6X6bKz6kTVczanIcsPaqtSTKk0Nf/view?usp=sharing))
2. Set infill to 10%

**Screenshot(s)**
Notice the fill below the small extension is at 10%? The extension breaks off easily.
[image]

Base layer for small extension.
[image]

Small extension only sitting on the 10% infill.
[image]

The provided STL file is for a small model. Units in mm.
[image]

On larger objects, the connection is solid. Top object given a solid ledge to sit on.
[image]

**Actual results**
Small connected object sits on infill, with no ledge. Must print at 100% to keep small connected object from breaking off.

**Expected results**
Expected detection of upper object, and the creation of a solid edge platform to sit on. If objects are too small, then switch to 100% infill in the layer just below the small object so they connect nicely.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
STL file is [here](https://drive.google.com/file/d/14gOY6X6bKz6kTVczanIcsPaqtSTKk0Nf/view?usp=sharing)
Gcode file is [here](https://drive.google.com/file/d/1HKJ_B1uye0pVceEsMM8t2CGFXjiAiS8U/view?usp=sharing)

smartavionics, 2020-03-06:
A workaround is to reduce the number of walls to 2 and so then you get skin layers at the join...

[image]

eskeyaar, 2020-03-06:
reducing the skin removal width also works.

aadhoc, 2020-03-07:
I provided a sample model as the minimum experiencing the problem. This is actually a single support from a quadcopter frame. Normally I'd have a screw hole in the top smaller cylinder, but the smaller cylinder always broke off. As a result I redesigned and filled in that smaller cylinder. Now I realize that the top was never really attached, which is why when solid it still breaks off.

My workaround: Due to the thin nature of my actual quadcopter frame, I can have it infill 100%, providing a full base to connect with.

This looks like something that could be detected/fixed in Cura, so I chose to share the issue.

Ghostkeeper, 2020-03-09:
The workaround that ekseyaar mentions is the intended one here. This piece of skin is thinner than the Skin Removal Width so it gets removed and replaced by infill. That is the trade-off that Skin Removal Width makes: productivity vs. reliability.

I've added a warning value now for Skin Removal Width to warn if it becomes larger than the wall thickness. If it is larger, then you could get a gap even for large structures. For smaller structures than twice the wall width, any removal width can be too big.