Ultimaker/Cura #7232: Cura Lulzbot [3.6.20] won't slice layer heights<0.14mm with support

OivindAlvestad, 2020-03-06:
**Application version**
Cura Lulzbot version 3.6.20

**Platform**
Windows 10

**Printer**
Lulzbot TAZ 6

**Reproduction steps**
1. Tried with several different models
2. Tried with several stock profiles, just changed the layer height to anything under 0.14mm
3. I can succesfully slice the same model with layer heights<0.14 without support. But the moment I try to add support, it won't let me slice.
4. Tried different infill amounts (0%/20%/50%100%)
5. Tried higher Initial Layer Height (0.3mm)

**Actual results**
Error message, and "Prepare" button is greyed out.
Error message: 
"Unable to Slice: 
Unable to slice with the current settings. The following settings have errors: Support Infill Layer Thickness"

**Expected results**
Slicing 

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
[CuraLulzbotSliceError.log](https://github.com/Ultimaker/Cura/files/4298033/CuraLulzbotSliceError.log)

**Additional information**
Sorry if this is out of your scope, and is only relevant for the Lulzbot team, I'll submit a bug report there as well.

mahtDFR, 2020-03-06:
We don't maintain Lulzbot Cura Edition, so better to bring it up with them.
According to your description I would say your support infill layer thickness is too high, so find that in your settings and change it to a lower value.

closed 2020-03-06 by mahtDFR

OivindAlvestad, 2020-03-06:
Ok, thank you:)
I thought the same thing, so I have tried adjusting the support infill layer thickness to the exact same as the layer height, but the problem remains. So I don't think that's the root cause...

---
Best Regards

Øivind Bremset Alvestad
Electrical Engineer
+47 95 782 781
oivind.alvestad@gmail.com

Ghostkeeper, 2020-03-09:
In our version, the value for Support Infill Layer Height needs to be between 1 and 8 layers (inclusive).