Ultimaker/Cura #7234: Allow to Edit the "Wipe Script"

thesec, 2020-03-06:
"Wipe Nozzle Between Layers" adds a automatically generated "wipe script" to the gcode.
Being able to edit that script, the same way as for the printer's and nozzle's start and end scripts would be helpful.

Ghostkeeper, 2020-03-09:
This script is integrated into CuraEngine. You can edit CuraEngine, but that requires compiling CuraEngine.

I don't think we'll make this easier in the near future. The "wipe script" is just as much a script as the "outer wall script" now.

Ghostkeeper, 2020-03-09:
It's somewhat ugly that this wiping always wipes horizontally to the X axis. It's really specific to the DesktopMetal type of printers that have their brushes on the side. In my opinion this sort of thing should not get merged into CuraEngine and so perhaps we should remove it from there.

closed 2020-03-09 by Ghostkeeper

thesec, 2020-03-09:
🥺