Ultimaker/Cura #7236: Cura 4.5.0 - Material specific nozzle/bed temps not transferred to Monitor screen for Preheat options (worked in all previous versions)

jimmieclark3, 2020-03-06:
**Application version**
Cura 4.5.0

**Platform**
Win 10

**Printer**
Ender 3 - 2 machines (stock ender 3, ender 3 with skr1.2 and microswiss)

**Reproduction steps**
1. Select material
2. Goto monitor screen
3. Preheat temps are blank (this used to autofill)

**Screenshot(s)**
[image]

**Actual results**
Have to manually enter these values every time

**Expected results**
They auto fill as previous

mahtDFR, 2020-03-09:
Duplicate of #7172
Fixed by #7178 

A fix will be available in 4.6. Thanks for the report.

closed 2020-03-09 by mahtDFR

DeezLBC, 2020-03-13:
I'm also having this issue. CR10s and Ender 3 on Win10

fieldOfView, 2020-03-13:
Everyone has this issue in Cura 4.5. It has already been reported and fixed for the next version of Cura.