Ultimaker/Cura #7238: Kossel printer selection missing from "Add Printer" List

solrac8874, 2020-03-07:
Kossel Mini or Kossel Pro printer profiles not available to select in "add non-networked printer" list even though there are def.json files for them in the Ultimaker Cura 4.5\resources\definitions folder.

In previous, Cura 3.x versions, thought I remember seeing Kossel Mini or Kossel Pro printers listed under "Other" printers sub-list.  But I don't see "Other" printers as a sub-list item either in Cure 4.x.

**Application version**
Cura 4.5

**Platform**
Windows 10 x64

**Printer**
(Which printer was selected in Cura?) Kossel Mini/Pro not available to select

**Reproduction steps**
1. Installed of Cura 4.5, then Opened Cura 4.5
2. Selected "Add Printer" Wizard
3. Expanded "Add non-networked printer" List
4. Searched for "Kossel Mini" or "Kossel Pro" printers - None Available to Select
5. Searched for "Other" printers to expand list - Not available to select
6. Do not want to select "Custom Printer - DeltaBot"
7. Want generic Kossel Mini printer profile linked to "kossel_mini.def.json" in Resources-Definitions

**Screenshot(s)**
[image]

**Actual results**
No Kossel Mini or Kossel Pro printer profiles available to select when trying to add printers.  Only comparable option is select Custom-Deltabot, but printer profile does not include "kossel.stl" platform reference as Kossel printer profiles do and does not match as closely with my Anycubic Kossel Linear printer.

**Expected results**
Expected to see Kossel Mini or Kossel Pro printer profiles within "add printers" selection list corresponding to the "kossel_mini.def.json" or "kossel_pro".def.json files found in C:\Program Files\Ultimaker Cura 4.5\resources\definitions folder.

All other .json printer files are represented in the "add printer selection list except Kossel printers.

**Project file**
N/A

**Log file**
N/A

**Additional information**
Same issue with past Cura 4.0 thru 4.4 version, I think. Previously was able to make Kossel printer profile available in the add printer list by copying, pasting, and renaming the "kossel_mini.def.json" to "anycubic_kossel_mini.def.json" into the C:\Program Files\Ultimaker Cura 4.5\resources\definitions folder.

Then I would be able to see Anycubic Kossel Mini printer available to select under Anycubic printers sub-list within the "add printer" list.  But this doesn't seem to work in Cura 4.5.

Searched Cura github for similar issue (open/closed), but could find a sufficient solution to my issue.

fieldOfView, 2020-03-07:
They are listed under the "Johann" subcategory.

solrac8874, 2020-03-07:
Thanks for you help.  I would have never guess it.  Now I feel a little silly.  Thanks again.

The Kossel Mini profile is a good starting profile for Anycubic Kossel Linear.  Only change in the printer settings needed is to change the Z-Max height to 300mm or so.  Works will enough for me.

It would be nice if Anycubic Kossel Linear profile could be added with the simple change from the Kossel Mini.

closed 2020-03-07 by solrac8874

AndreasGerken, 2020-04-30:
This topic is very helpful as nobody would ever guess to search for "Johann" in the long list of printers. The settings are not even accurate and have to be modified before using it. It would be great, if a profile would be added with the proper name.

fieldOfView, 2020-05-01:
It would be great if someone submitted such a profile.

Ghostkeeper, 2020-05-06:
The Kossel Mini profile in Cura is referring to this one by Johann: https://github.com/jcrocholl/kossel
There are multiple printers with the same name here, it seems. Probably based off each other.