Ultimaker/Cura #7239: Buildplate glas breaks. (When printing big parts with CPE+)

PvD77, 2020-03-07:
**Buildplate glas failure.**
When a big 3D print is printed on the glass with CPE+:
The 3Dprint sticks very good at the glass. 

Bud when ready: the glas will break because the 3D printed part shrinks more than the glass, when cooling down to 20 degrees.

Easy solution:
Add setting to keep the buildplate at a fixed temperature when the 3D printing is ready.
Probably 40 degrees is good. (make the value adjustable)
At this temperature the glass is one piece and in good condition.
And the product is removable. (Use alcohol around it.)

Used nozzle temperature (on UM3): 260 degrees (recommended)
Buildplate temperature: 110 degrees. (recommended)

mpsdskd, 2020-03-08:
Potentially, applying a thin layer of glue stick glue onto the glass could reduce the adhesion between part & build plate enough to not destroy the glass bed (you can find this solution online, too).
I did not test this with CPE+

PvD77, 2020-03-09:
Thanks mpsdskd,

I'am aware of those tricks. 
Working with clean glass and with maximum adhesion is very plesant to work with. 
When I reduce the adhesion, the 3D printed product will be more sensitive to get free from the buildpalte.

Ghostkeeper, 2020-03-10:
I think Cura shouldn't keep the build plate warm after the print, because:
* Most firmware will cool down the build plate explicitly after the print.
* If it doesn't, this is a risk of excessive energy usage.

I'd rather have the firmware give an option to temporarily heat up the build plate. Or you can use Cura's pre-heat functionality which also heats up the build plate to a limit of 15 minutes.

mpsdskd, 2020-03-11:
How about putting a build plate heat command into the End Gcode?
M140 S _temperature_
Not sure about your printer's behavior, if it stops all heaters if the last GCode command finished - maybe an additional wit command to keep it heating?
G4 S _wait time in s_

PvD77, 2020-03-15:
> I think Cura shouldn't keep the build plate warm after the print, because:
>
> * Most firmware will cool down the build plate explicitly after the print.
> * If it doesn't, this is a risk of excessive [...]

The glas plates brakes when the buildplate temperature is low. (Around 20 degrees.)
Even when no human has touched it.

PvD77, 2020-03-15:
> How about putting a build plate heat command into the End Gcode?
> M140 S _temperature_
> Not sure about your printer's behavior, if it stops all heaters if the last GCode command finished - maybe an additional wit command to keep it heating?
> G4 S _wait time in s_

That is a good idea!
:)

Ghostkeeper, 2020-03-17:
We'll reject this issue as the problem is really printer-specific and can be solved in the end g-code, which is arguably intended to solve such printer-specific issues.

closed 2020-03-17 by Ghostkeeper