Ultimaker/Cura #7240: Infill fuse lines

Stormwing01, 2020-03-07:
**Application version**
4.5 and previous versions

**Platform**
Windows 7

**Printer**
Wanhao I3 mini

**Reproduction steps**

I am unsure why it happens, but it only seems to happen when I activate tree support and there are many spots Tree support has to cover. it seems to happen at or near the top of tree supports.

the screenshot examples are at the bottom of the model.

I have disabled normal supports whenever I use tree supports.

**Screenshot(s)**

[image]

[image]

[image]

**Actual results**
It creates a few layers that goes directly through the model itself and thus fuse the model to the support. it also creates a grid of support material that does not actually support anything. (you can see that in the middle of the model, for example.)

**Expected results**
The screenshot shows what happens. 

**Project file**
[New WinRAR ZIP archive.zip](https://github.com/Ultimaker/Cura/files/4301824/New.WinRAR.ZIP.archive.zip)

**Log file**
I cannot find the log file even when I followed the directions. 

**Additional information**
None, other than it's been there for a few versions. I think I first saw it in version 4.0. but I am unsure.

Ghostkeeper, 2020-03-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/5223.

closed 2020-03-10 by Ghostkeeper

nallath, 2020-07-01:
This has been fixed for 4.7