Ultimaker/Cura #7241: Problem with infill only mesh

mpsdskd, 2020-03-08:
**Application version**
4.4 & 4.5 tested.

**Platform**
Linux (Kubuntu 19.04).

**Printer**
Custom FFF.

**Reproduction steps**
So this is quite an edge case & might also be user error on my part.

Idea:
- I want to print lightweight parts on my printer. The solution to do so, which I found the easiest, is to have the whole part modelled solid in CAD, use this as shell only (infill 0%) and use a second model as the internal structure using it in Cura as an "Infill only" model.

If I do so (see screenshot &  [HERE](https://github.com/mpsdskd/3D-Print-Plane/blob/master/3d-test/13_fuselage_not_nose.3mf)) it does not work as expected.

**Screenshot(s)**
[image]

**Actual results**
The second model (internal structure) is shown in the preview tab with the color red for shell & the infill setting in the print settings area are not shown at all (infill density = 0% with main model).

**Expected results**
Infill settings should be shown for the infill only model & infill model should be shown as infill in preview.

**Project file**
[https://github.com/mpsdskd/3D-Print-Plane/blob/master/3d-test/13_fuselage_not_nose.3mf[(https://github.com/mpsdskd/3D-Print-Plane/blob/master/3d-test/13_fuselage_not_nose.3mf)

**Additional information**
I think the "infill only" model is not recognized as infill by cura, if it only contains thin parts at least, thus creating this behavior.

Yes, this is quite an edge case & I can understand prioritizing it low.
Also workarounds for my problem are welcome :)

smartavionics, 2020-03-10:
Try this...

1 - Delete or move out of the way the wacky large object 1.
2 - Move your solid object so that it is located at 0,0,0.
3 - Make a copy of that object and set it to be an infill mesh.
4 - Move the infill mesh to 0,0,0.
5 - Set the desired infill settings on the infill mesh and slice!

[image]

The blue blotches are just there because the infill mesh is currently selected.

smartavionics, 2020-03-10:
I deleted my comment as I realise that I had completely misunderstood the problem, sorry.

smartavionics, 2020-03-10:
OK, second attempt at being useful....

Here's some settings for your infill object that appear to generate walls in what looks like the right places...

[image]

mpsdskd, 2020-03-11:
Yes, those settings create _something_ where both meshes overlap. 
This is also the case in my example image.

Judging from the color in preview it is not infill, though (red=shell). 

Maybe I'm using the function wrong?
I try to have the main object for its shell and I want infill only where the main mesh and the infill mesh object overlap. 
With the settings provided in my cura file it does create something in the overlapping regions. _But_ it is not shown as infill in preview and is does not care about infill settings (I'm mainly interested in the infill/shell overlap setting). 

If you look closely into both your and my screenshot the infill mesh wall and the main object shell do not realy overlap and this is also the case for a real print: Infill and wall do not adhere to one another.

Ghostkeeper, 2020-03-11:
I think there's a bit of a misunderstanding of how these infill meshes work.

An infill mesh is basically just a mesh that gets clipped to the volume of your infill. Commonly this looks something like this (ignore the weird blend of English and Dutch, was testing something):
[image]
The cylinder is an infill mesh and it gets clipped to the infill volume of the cube. The cylinder still has walls, so those are clipped to the infill volume of the cube too.

Your print doesn't have any infill (it's set to 0%). You do have infill meshes, and those have 1 wall just as the rest of your print. So you're going to see the walls of this infill mesh being printed just as normal walls. Because they are. They will use the outer wall settings too, rather than the infill settings (so it will print slower and such).

You might be able to get it to print correctly by setting the number of walls and top/bottom thickness to 0 on that infill mesh and using concentric infill at 100%. However I don't think that the speed is your real problem here. I think your real problem is that the infill then has no overlap with your normal walls, and I don't think Cura has a solution for you in that regard.

mpsdskd, 2020-03-11:
@Ghostkeeper 
Thanks for this input, so my understanding of "Infill only" was completely wrong.
What might my problem, is settings the shell width of the infill model to 0 lines, as you said, I will try.
Thanks a lot!

I still think the infill settings should not be hidden, if the infill of the main model is set to 0 percent, but a second mesh contains infill with a density != 0.

closed 2020-03-11 by mpsdskd

Ghostkeeper, 2020-03-13:
They won't get hidden from the per-object settings panel where the density is set higher, are they?

mpsdskd, 2020-03-14:
No, but if the general infill density is set to 0% the infill settings are hidden, although they affect the second object (e.g. infill pattern), where the infill density is set to != 0 with the per model settings. Yes, I could still change infill settings with more of the per model settings.